raomin/ESPAltherma #83: error 22 - bat vol error check hardware

crounauer, 2022-02-21:
I have ordered 2 M5Stick C and both have an error code "error 22 - bat vol error check hardware" when first plugged into usb-C. I have tried charging them overnight but still the same error appears. Does anyone have a suggestion to fix this error?
Thanks ;)

qris, 2022-02-21:
Have you seen [this article](https://community.m5stack.com/topic/2130/error-22-bat-vol-error/11)?

crounauer, 2022-02-22:
Hi qris,
Yes, I've tried all those options.
Thanks for your reply.
Cheers!

raomin, 2022-02-22:
Hello @crounauer 
I'm sorry to hear that, but it's unlikely that this is the best place to find help for that.
But is it preventing you from programming the esp32?

crounauer, 2022-02-23:
Hi Roamin,
Yes, it's not possible to get past the error message.
I'll order from another shop, maybe it's old stock?

crounauer, 2022-02-23:
Would this ESP32 board work? => ESP32 Development Board BLUETOOTH WIFI - ESP-WROOM-32 Arduino Raspberry PI IoT

raomin, 2022-02-23:
Have you tried to open the m5 and disconnect the battery?
If still not working, yes any esp32 would work. Its just that m5 has few advantages (magnetic, screen...)
You can get a m5stickc plus, it shall be fresher...

crounauer, 2022-02-23:
That's a great idea, why didn't I think of that? Nothing to loose, I'll give that a try tomorrow. Thanks.

raomin, 2022-11-22:
I'll close this one, I hope you solved your m5 issue...

closed 2022-11-22 by raomin