raomin/ESPAltherma #85: Technician Access Code for smart grid

vekexasia, 2022-03-03:
I can't really set the Smart Grid functionalities due to the fact that the menu is not shown to my user access level. 

I tried 0000 (which gives me "OK") but shows nothing different, 1234 - 5678 do not work.

Any idea on how to get mine?

closed 2022-03-03 by vekexasia

raomin, 2022-03-04:
solved your issue?

vekexasia, 2022-03-04:
Yes found the code by asking the installer :)

-- 

Andrea Baccega
[image: ***@***.***
[image: ***@***.***
[image: Personal site] <http://www.andreabaccega.com/> [image: Linkedin
Profile] <http://it.linkedin.com/in/andreabaccega> [image: Facebook]
<https://www.facebook.com/andrea.baccega> [image: Google+]
<https://plus.google.com/109217393200753135791> [image: Twitter]
<http://twitter.com/veke87> [image: StackOverflow]
<http://stackoverflow.com/users/314407/andrea-baccega> [image: GitHub]
<https://github.com/vekexasia>