raomin/ESPAltherma #86: Advise definition for ehvh08s18ea9w please

mgx0, 2022-03-06:
Hello

thanks for this genius project, I really appreciate it. Got it working but I would like to get it right and need help. 

I have DAIKIN EHVH08s18ea9w pump. Could  you please advise which definitions to use?

is there any chance to get energy meter readings as well (the ones available on the Daikin menu on the unit and within the app)?

Thanks a lot again!!

mgx0, 2022-03-06:
I have a feeling you will recommend the same as for #31 but anyway, I’d appreciate a reply 😊

raomin, 2022-03-07:
Hi @mgx0,
Have a look at https://github.com/raomin/ESPAltherma/wiki/Choosing-the-right-definition-file
I would go for ALTHERMA(LT_DA_04-08KW).h

closed 2022-03-07 by raomin

mgx0, 2022-03-07:
Thanks @raomin , in the meantime i went with what you suggested. Looks fine for me. Can you update the table with my model? So nobody else asks the same again 😊

thanks a lot for your work

raomin, 2022-03-10:
@mgx0 , you can edit the page yourself! :)
Just go there on the page https://github.com/raomin/ESPAltherma/wiki/Choosing-the-right-definition-file and click edit on the top right!
Thanks!

mgx0, 2022-03-10:
> @mgx0 , you can edit the page yourself! :)
> Just go there on the page https://github.com/raomin/ESPAltherma/wiki/Choosing-the-right-definition-file and click edit on the top right!
> Thanks!

Ooh right, thanks :-)