raomin/ESPAltherma #88: unknomn value

Krinou3000, 2022-03-18:
Hi and great job @raomin !

I install an M5StickC Plus on my EHVH04S18CB.

But I have often some `unavailable` values on the AlthermaSensors entity and this cause at the same time `unknown` values for the sensor.templates I have create.

I don't know exactly why. Sometimes I don't have this issue for several hours .
Wifi signal seems good with router (WifiRSSI -58dBm).

[image]

raomin, 2022-03-19:
Hi @Krinou3000 
That's probably network/wifi connectivity issue. We you observe this behavior, do you manage to ping the m5?

Krinou3000, 2022-03-20:
Yes, you right. I test with the esp a little more outside the case and I put the antennas of the router straight.
 I no longer have any data reception cut off.

Krinou3000, 2022-03-20:
Now, I have this ping result. I don't know if it's good or not for the M5StickC

```
Envoi d’une requête 'Ping'  192.168.2.242 avec 32 octets de données :
Réponse de 192.168.2.242 : octets=32 temps=49 ms TTL=255
Réponse de 192.168.2.242 : octets=32 temps=356 ms TTL=255
Réponse de 192.168.2.242 : octets=32 temps=256 ms TTL=255
Réponse de 192.168.2.242 : octets=32 temps=81 ms TTL=255

Statistiques Ping pour 192.168.2.242:
    Paquets : envoyés = 4, reçus = 4, perdus = 0 (perte 0%),
Durée approximative des boucles en millisecondes :
    Minimum = 49ms, Maximum = 356ms, Moyenne = 185ms
```

raomin, 2022-03-23:
Your pings speed jitters a lot (from 49ms to 356), that's not good for your network.
But this might be "good enough" for ESPAltherma. If you still have unknown values, you can also leave the M5 outside the heatpump... Also, personally, I did not replace the metallic plate to avoid faraday cage effect...
I close the issue, reopen it if you have news!

closed 2022-03-23 by raomin