raomin/ESPAltherma #91: Rotex HPSU 508 Compact  ESP32

Micha031982, 2022-03-24:
Hi,

i tried different things to get this working but failed.

Setup:
Rotex HPSU 508 Compact
used can J13 anfd Ground from here
[image]

 ESP32 with usb power (ESP-32 Dev Kit C V2)
Pin 16/17

Setup with the same like here in a other Post
#include "def/ALTHERMA(HPSU6_ULTRA).h"
and the same parameters like in the other poste

Tries:
different Ground on ESP
Different directions of can
ground to all grounds of the ESP
2 sets of jumper cable
second ESP

What i get:
[image]
[image]

Any idea what i can try additionaly?

raomin, 2022-03-24:
Do I read you well? You connected ESPAltherma on the CAN bus??
Err... ESPAltherma is supposed to be connected to the serial port X10A, not the CAN bus...

Micha031982, 2022-03-24:
Thanks for the quick answer
Works now (had mixed up this manual with those for FHEM)

closed 2022-03-24 by Micha031982