raomin/ESPAltherma #92: How can i automate thermostat and SG relay?

colseven, 2022-03-25:
Hello again :)

last question, i installed the relay for the thermostat and smart grid and can also control them - it all works. but if i build an automation then i can't select and automate the devices. where is my mistake??

Thank you very much.

[image]
[image]
[image]

colseven, 2022-03-25:
solution

`type: turn_on
device_id: Altherma
entity_id: switch.altherma
domain: switch`

`type: turn_on
device_id: SG Altherma
entity_id: switch.sg_altherma
domain: switch`

raomin, 2022-03-25:
Yup! You got it! :)

closed 2022-03-25 by raomin