raomin/ESPAltherma #94: Altherma 3 EABX16DF6V

coralblack12, 2022-04-03:
I have installed the Altherma 3 Medium Temp. the motherboard seems not have the X10A  connector, but  in the other hand there's 1 USB, 2 network port and port similar to a memory card slot. 
How can I install my ESP32 card?
Thanks a lot in advance!

See attached image 
[image]
[image]

raomin, 2022-04-03:
You're not looking at the right board. This one controls the UI. 
You need to open the metal plate against this one is laying.

raomin, 2022-04-03:
BTW these are not network ports but RJ45 supporting CAN protocol. I'll explore this one day....

coralblack12, 2022-04-03:
Thanks a lot. I've found the connector, buy the way the ESP32 card seems not powered on. I cannot see it connected to the network.
The cables are connected as you described, I have to check the power supplied by the first pin of the connector. Or do you have another suggestion?

coralblack12, 2022-04-04:
Many thanks, I've found the port. Another question. I'm going to use an ESP32 card. Do you think it's mandatory to have a Raspberry/PC always connected to collect data or the card could store data and once per day/week I can download it via Platformio?
Since I don't have a Raspberry neither a server always on.

raomin, 2022-04-04:
I'm considering to build a data archive/reporting web interface but not for the short term.
So right now, yes, you need at least an MQTT server that will archive the data. You could consider using an online service like Adafruit IO, azure iot, etc... This would need a bit of code adaptation though...

closed 2022-04-05 by raomin