raomin/ESPAltherma #97: Q: use ESPAltherma from within ESPHome

stomko11, 2022-04-08:
As stated in title: is there a way to use ESPAltherma from within ESPHome, maintain code updated and wirelessly update it from within ESPHome (HomeAssistant?)

raomin, 2022-04-22:
At the moment, there is no plan to develop a ESPHome component for ESPAltherma.
This is probably fairly doable so if someone is motivated, this can happen!

jjohnsen, 2022-10-01:
I've hacked togheter ESPAltherma and ESPHome, and had it running for about a month. Its quite rough, but if there is interest I might clean it up a bit and poste the code.

stskrp, 2022-10-08:
@jjohnsen would be really cool to see implementation with ESPHome :)

remycrochon, 2022-12-01:
Hello, I am also interested in this subject!

tomaszsluszniak, 2023-01-08:
> I've hacked togheter ESPAltherma and ESPHome, and had it running for about a month. Its quite rough, but if there is interest I might clean it up a bit and poste the code.

Hi, could you please share the code?

chris24walsh, 2023-02-20:
> I've hacked togheter ESPAltherma and ESPHome, and had it running for about a month. Its quite rough, but if there is interest I might clean it up a bit and poste the code.

@jjohnsen please do share, even if rough. Would be very helpful

gervaso, 2023-08-21:
@jjohnsen  I'm interested in this implementation as well, even if rough it would be a starting point.

divenal, 2025-12-04:
I'm just trying to understand this suggestion... is the proposal that esphome would be running on the esp32 attached to the altherma, then the code from this project would be embedded as an esphome sensor ? This way, esphome protocols would be used to deliver the sensor data, rather than mqtt ?

I've been mulling over the possibility of using protocols other than mqtt (which is a bit heavyweight for my needs) to deliver sensor data.

jjohnsen, 2025-12-24:
Ok, I’ve had a look. The code is 3 years old and it’s rough.

However, it has been running without problems for the last few years.

I’m going to share it with @nicx. If anyone else is interested, just give me a message. Perhaps it’s possible to turn it into something useful for others as well.

[image]

[image]

jjohnsen, 2025-12-24:
> Would like to use it as a starting point as well. Seems quite a few people would be interested to get a EspHome version 😉

I’ve added you. Having it in ESPHome simplifies things . It’s just a shame I never had time to clean up the code after getting it working on my specific installation.

divenal, 2025-12-24:
@jjohnsen may I see it too, please.

I've actually been playing with my own implementation - see https://github.com/raomin/ESPAltherma/discussions/252#discussioncomment-15328434

And FWIW there's another ESPHome implementation mentioned earlier in that discussion: https://github.com/raomin/ESPAltherma/discussions/252#discussioncomment-11704356

divenal, 2025-12-24:
The thing I had been concerned about was blocking while waiting for the response - doesn't that lock out the whole ESPHome loop ? Not an issue if the device is dedicated to speaking to the Altherma, but one reason for using ESPHome is to have additional sensors, and I assume that not yielding to the loop promptly would be bad.

ie the reader would need to be a state machine that advances as characters arrive, rather than just blocking to read the whole response (or worse, waiting for the full timeout period).

MaBeniu, 2025-12-25:
@jjohnsen , It would be awesome if would add me as well.

jjohnsen, 2026-04-06:
@divenal finally had time to look into this🙄, and you were quite right.

Using the ESPHome runtime_stats component, the original runtime was quite bad, averaging 288 ms:

```
Component Runtime Statistics
 Total stats (since boot): 5 active components
  altherma_hub: count=456, avg=288.581ms, max=375.53ms, total=131593.0ms
```

Switching to a non-blocking, state machine implemntation improved things quite a lot:
```
Component Runtime Statistics
 Total stats (since boot): 5 active components
  altherma_hub: count=104913, avg=0.020ms, max=35.72ms, total=2138.9ms

  wifi: count=104885, avg=0.006ms, max=9.66ms, total=610.2ms
  api: count=104885, avg=0.006ms, max=5.32ms, total=609.2ms
  esphome.ota: count=104885, avg=0.002ms, max=0.28ms, total=241.0ms
  preferences: count=14, avg=0.638ms, max=8.44ms, total=8.9ms
```

This aligns much better with ESPHome’s guidelines and should make it safer to run alongside other components.
Thanks for pointing it out.

henrykuijpers, 2026-08-25:
@jjohnsen Have you thought of being able to import historic data from ESPAltherma into esphome-altherma? For those users with installations that have been running for years and years and would like to switch to esphome-altherma, without loosing all their statistics?

jjohnsen, 2026-09-03:
@henrykuijpers I haven’t investigated this. I've been running ESPHome Altherma for several years, so I do not have an existing ESPAltherma history database to test with.

It should be possible, since the historical data is stored by Home Assistant rather than ESPAltherma itself.

If there is interest, I can investigate it. You open a feature request in the ESPHome Altherma repository. That would give us somewhere to document and test the migration approach.

raomin, 2026-10-23:
Moving to Discussions. ESPAltherma 2 stays its own firmware (the web installer and the runtime setup were the point), and the ESPHome ports by others keep living; the thread is the place to compare them.