raomin/ESPAltherma #103: Home assistant Smart Grid input automation

RodEzIO, 2022-04-24:
Hello,

Does anyone has an automation example to share for integrating smart grid feature by using home assistant?

I am looking for something that starts the SG mode when the PV production is above X kw. 

Thanks

raomin, 2022-04-24:
@andreapier, @maps2002, @colseven could possibly help.
I don't have PV production.

colseven, 2022-04-25:
Hello,
I have Solaredge/Home Assistant and I exceed my hot water target value from 48°C to 54°C with a pv excess.

- id: '1648201081271'
  alias: Warmwasser auf 55°C bei 0,5 kW Strom nach 5 Minuten export EIN
  description: ''
  trigger:
  - platform: numeric_state
    entity_id: sensor.solaredge_m1_ac_power
    for:
      hours: 0
      minutes: 5
      seconds: 0
    above: '500'
  condition: []
  action:
  - type: turn_on
    device_id: SG Altherma
    entity_id: switch.sg_altherma
    domain: switch
  mode: single
- id: '1648201153233'
  alias: Warmwasser auf 55°C bei 0,2 kW Strom nach 5 Minuten export AUS
  description: ''
  trigger:
  - platform: numeric_state
    entity_id: sensor.solaredge_m1_ac_power
    for:
      hours: 0
      minutes: 5
      seconds: 0
    below: '200'
  condition: []
  action:
  - type: turn_off
    device_id: SG Altherma
    entity_id: switch.sg_altherma
    domain: switch
  mode: single

maps2002, 2022-04-25:
Hi @RodEzIO ,

I am using a similar automation as @colseven , only based on the current solar power production.
I am not able to read any values from my heat pump model, so also cannot "complicate" the automation with any heat pump state values :)

raomin, 2022-04-30:
I guess this one can be closed now.

closed 2022-04-30 by raomin