raomin/ESPAltherma #108: Powerconsumption of Backup Heater BUH

niekveldhuizen, 2022-05-13:
Hi,

Now when we get in to the higher temperatures outside i see that my Daikin is using its BUH heater for heating DHW. It seems to be related to the outside temperature.

1. When only BUH is active, the Daikin Operation Mode does not state "Heating"
2. Primary (or secondary) inverte current is not giving any value (because internal heating element is used

Problem 1 is the most pressing, I would like to know how much energy is being used by the internal element (BUH?)

KalaNagHTD, 2022-05-13:
You‘ll find „BUH Step1“ and „BUH Step2“ in the data. The BUH can deliver 3, 6 and 9kW. Unfortunately I don‘t know which is Step1 and Step2 and you have to find out (or anybody else here knows about).

niekveldhuizen, 2022-05-13:
Oké I can enable those values. Do they return a amperage or wattage?

what I want to create is that I can count the energy spend in an continuous kW counter. Is that possible?

I also trying to figure out why it is using the BUH at all. I don’t want it to :)

m-reuter, 2022-05-13:
One way to avoid BUH is to take it out of your configuration completely (then it will never be used neither for heating nor for warm water). This should be OK, if you have a well insulated house and not too cold temperatures in the winter (if you heat pump is not strong enough, you can easily add BUH back in). (Settings->external source->config ext. heat source , again not sure about the exact english menu names). 

Another approach to switch off BUH during warm water only by setting the time when BUH can be used to longer than the WW time. This is the Wait time for external BUH (not sure what it is called in english exactly), set it to 90 mins and the max WW load time to 60. Then it will stop after 60 min to make hot water and never use the BUH. 

Note that my model also wants to use the BUH when the outside thermostat shows really high degrees (around 30 celsius, e.g. if in direct sun). If that is the case, it will probably try to use BUH anyway, unless you remove it completely from the configuration. In the later case it will not make warm water and wait until it is colder outside. Best to put heat pump (or at least the outside thermostat) in the shade. 

I wrote a few notes on my settings (HPSU Compact Ultra / Altherma 3 R ECH2O) here (but in German): 
 https://m-reuter.github.io/Rotex-Daikin-WP/hpsu-compact.html

niekveldhuizen, 2022-05-19:
@KalaNagHTD 
Thanks
I will try to enable those values then. But do you get en "on" "off" feature of something regarding the power being used? 

@m-reuter 
Thanks
This is the first spring and summer with the heatpump. So all is new. Last winter the BUH was used for defrost (i think?) so i think in wintertime i can not shut it down complete.   
My outsite temperature feeler is direct sunlight and in a special enclosure with the outdoor unit itseld,  so even if it was 26/27 yesterday outside, the unit measured 35 and then it does not use compressor, only BUH. I asked the company that installed the pump to correct this. It should be possible with an external feeler. 

I think i have similar problems in wintertime. The outdoor unit is getting cold from doing its thing, and the feeler is right next to in the enclosure, it wil measure colder temperatures that it actualy is. I think that is why it sometime uses a defrost cycle when it should not me neccesary.

I will have a look at you note's. I have a different model, but same pricipals apply

alexdelli, 2022-05-24:
I switch to a shelly EM to calculate consumption. It is more easy and I found different than estimation from AMP x VOLTAGE

niekveldhuizen, 2022-05-24:
@alexdelli,

I have been looking into the Shelly EM, i need an Shelly 3EM. I can not find any installation guides for use to 1 particualt 3fase group in my hous. Only how to install for the incomming line in the house. How do you have it installed?

mgx0, 2022-06-05:
Back to topic: anyone seen this?
[image]

Seems Altherma 3 compressor is not being used for lwt setpoint below 25. 

does anyone have this information? This picture is nowhere to be found.

mgx0, 2022-06-05:
Update, managed to find the doc: https://www.kupklimatyzator.pl/file_store/ERGA04-08EV_EEDEN20_Data%20books_English.pdf

really if you don’t use thermostat and rely only on LWT and weather curve, you can end up with extreme electricity costs like I did. 

Do not use LWT below 26 degrees or your Backup Heater will fire up

sahni2021, 2022-12-06:
> You‘ll find „BUH Step1“ and „BUH Step2“ in the data. The BUH can deliver 3, 6 and 9kW. Unfortunately I don‘t know which is Step1 and Step2 and you have to find out (or anybody else here knows about).

Today I noticed that something is not quite right with my setup, because the heatpump was using the backup heater while none of the attributes in the MQTT message gave a hint about it. I have attached the json lines as well as photos taken at the same time of the event. So I wonder why BUH Step1/2 was OFF when the backup heater was actually ON. Any advice welcome.

{
  "Operation Mode": "Heating",
  "Defrost Operation": "OFF",
  "Outdoor air temp.": 4.5,
  "Discharge pipe temp.": 72,
  "O/U Heat Exch. Mid-temp.": 1.5,
  "INV primary current (A)": 9.5,
  "INV secondary current (A)": 12.7,
  "Voltage (V)": 237,
  "INV frequency (rps)": 46,
  "Fan 1 (10 rpm)": 67,
  "Expansion valve 1 (pls)": 238,
  "I/U operation mode": "DHW",
  "Freeze Protection": "OFF",
  "Silent Mode": "OFF",
  "DHW setpoint": 51,
  "LW setpoint (main)": 50,
  "3way valve(On:DHW_Off:Space)": "ON",
  "BUH Step1": "OFF",
  "BUH Step2": "OFF",
  "Outlet Water Heat Exch. Temp. (R1T)": 54,
  "Outlet Water BUH Temp. (R2T)": 50.6,
  "Refrig. Temp. liquid side (R3T)": 50.1,
  "Inlet water temp.(R4T)": 50.1,
  "DHW tank temp. (R5T)": 50,
  "Powerful DHW Operation. ON/OFF": "OFF",
  "Circulation pump operation": "OFF",
  "Flow sensor (l/min)": 19.6,
  "Water pump signal (0:max-100:stop)": 49,
  "Pressure sensor": 33.6,
  "M5VIN": "4.95V",
  "M5AmpIn": "95mA",
  "M5BatV": "4.13V",
  "M5BatCur": "0mA",
  "M5BatPwr": "0mW",
  "WifiRSSI": "-44dBm"
}
[image]
[image]
[image]

KalaNagHTD, 2022-12-06:
The correlation between the two steps and the power level is defined in the setting of your Altherma. For either step you can chose 3, 6 or 9 kW.

rayxxxxx, 2023-01-06:
I’ve got the same issue: buh step 1 and buh step 2 are OFF when the backup heater is ON.