raomin/ESPAltherma #109: Read daily consumption

Daikineasy, 2022-05-15:
I would read the daily consumption and thermal production. Is there a way to read these values from the pump menu or to calculate it?

Arnold-n, 2022-05-15:
The main controller menu enables to read these manually. On my system the electricity consumption counters (resolution 1kWh) are updated exactly once every hour (an external E-meter would do much better); the energy produced counters (resolution also 1kWh) are updated as soon as an extra kWh is produced (and an estimate made based on water temperatures and water flow would do better). I poll these counters every minute via the P1/P2 (thermostat) interface.

Daikineasy, 2022-05-16:
Hi Arnold,
With your solution I have to mount another ESP card to the serial port? At the moment I'm using an ESP32 which is connected to my Home Assistant installed to a Server PC. Do you think I can collect all the info via Home Assistant for both kind of communication?

Arnold-n, 2022-05-16:
Hi Daikineasy,
Indeed, the P1P2-ESP design has its own ESP8266 and is a separate circuit connected to the external thermostat/controller P1/P2 cable. It will become a 2nd entity in Home Assistant next to ESPAltherma.

raomin, 2022-05-16:
Hi @Daikineasy,
If your heat pump supports it (most of them do), the current consumption is reported in `INV primary current (A)`
You can use Home Assistant's [Riemann sum integration](https://www.home-assistant.io/integrations/integration/) to keep track of the consumption overtime.

The thermal production can also be calculated, that's the numerator of the COP value. See calculation on the Readme page.

closed 2022-05-31 by raomin