raomin/ESPAltherma #112: OTA update m5stick plus

jangarga, 2022-05-29:
try to configure m5stick to update firmware by OTA, but during firmware upload system show below error:
[image]

can you help me?

munzili, 2022-05-29:
try to replace the  upload_port in platformio.ini with the IP of the ESP32

jangarga, 2022-05-29:
already try, but have still problem

jangarga, 2022-05-29:
[image]

munzili, 2022-05-29:
Are you sure its the correct ip? Can you ping it?

raomin, 2022-05-29:
Also, as the ESP is connecting subsequently to the pc, one common issue is that the connection will be blocked by the firewall on the PC. Make sure it's deactivated for the time of the update.

jangarga, 2022-05-30:
> Also, as the ESP is connecting subsequently to the pc, one common issue is that the connection will be blocked by the firewall on the PC. Make sure it's deactivated for the time of the update.

With your instruction, working fine, thanks for support

closed 2022-05-30 by jangarga