raomin/ESPAltherma #113: State of the sensor.althermasensors is allways in unknown (none) state

andreytse, 2022-06-03:
Is it possible to change or override forever state of the sensor.althermasensors?
It is allways in unknown (none) state

upd.
Maybe it will be good to know about esp32 or m5 is still reading something from Altherma. I mean if any attribute was readed from heating pump so sensor state = On, if no one register was readed, sensor state = off

Gollam, 2022-06-06:
I adjusted it manually to Online in the code, but I agree, this would be more meaningful.

munzili, 2022-06-27:
I will take a look into this after i finished the WebUI feature pullrequest.

raomin, 2022-11-19:
Finally found some time to fix it!
[image]
It's now set to the main state. Online/Offline.

closed 2022-11-19 by raomin