raomin/ESPAltherma #114: OutdoorAirTemp_R1T

mgx0, 2022-06-06:
Hello,

I have connected a thermostat (Madoka), at the same time ONECTA App requested a firmware update (which was not related to madoka, I have postponed it multiple times), so I have installed it.

now I am getting incorrect readings here: 
{0x20,0,105,2,1,"OutdoorAirTemp_R1T"} <- shows 11.5 instead of 20 degrees Celsius. On the HeatPump screen there is 20 degrees outside displayed

how can I make a "debug" version of ESPAltherma to see which register now contains the OutdoorAirTemp_R1T temperature please?

I am not asking anyone to fix it for me, I want to find it myself but I need to show a good way to do it

thanks

raomin, 2022-06-06:
That's super weird... You did not change anything on your ESPAltherma? 
What happens if you disconnect the thermostat?

mgx0, 2022-06-06:
Nope nothing has changed, “only” firmware update

Thermostat was connected 2 weeks after this behaviour has started, so it does not have much to do with this. 

Other values seem to be reported okay. 

I have noticed one thing: when there is no heating at all, the Outdoor temp is not reported properly. But if there is DHW heating in progress, outdoor temp starts to be reported okay. I’ll attach an example screenshot

mgx0, 2022-06-06:
I'm sorry I was wrong. It is not the firmware update that broke this. Looking at attached screenshot, the date when the outdoor temperature started to be reported incorrectly seems to be the date I turned off the floor heating completely

[image]
I

mgx0, 2022-06-07:
Yes I have slept it over and it makes sense. Fair point, thanks

closed 2022-06-09 by mgx0