raomin/ESPAltherma #115: wrong CRC on EHVX08S26CB9W model

mauromorello, 2022-06-12:
Hi everyone, I'm having some issues to figure out what's the problem with my pump.... I followed all instructions I found here but the best i can achieve is from "Timeout" to "Wrong CRC on registry ...." I'm not good at programming, so the best I managed to do is to put on mqtt also the message "Calculated 0xff but got 0x 0"; I think the serial connection is working but the protocol is not correct. 
Does someone have some suggestions? Thanks a lot.

munzili, 2022-06-27:
Well, can you give more information? Photos how its connected to the X10A port and your ESP32. Are you using a M5StickC? Can you post the display output?

raomin, 2022-06-27:
Thanks @munzili.
@mauromorello Most probably, this is a communication error. Please check all you wires expetially the GND - GND connection. Even if you connected your M5 with a USB powersupply, you need to additionally connect the GND of the M5 to the GND of the Heat pump.

mauromorello, 2022-07-05:
OK, thanks a lot and sorry for my delay; 
1) I'm using a LOLIN32 for this project so I don't have display information
[image]
2) I use the altherma +4,7V to power LOLIN32, by the battery connector, and i check that **the - contact is connected to GND on the board**
[image]
3) If I run without TX and RX connected my MQTT message is a "TIME OUT"
[image]
4) when I connect TX and RX i have the CRC failure message
[image]

I'm not a PRO but I try putting the calculated CRC value also in MQTT log, and I see that that CRC is 0, as if try to communicate but without understand what my Altherma is saying.

Before go deeper and trying to connect with Daikin software maybe someone have some other suggestion

thanks a lot (espacially @raomin for this useful project)

mauromorello, 2022-07-16:
SOLVED;

i Write here just because of this weird behavior;

it was basically my fault, i used a damaged dupont connector on RX side; it was touching/nontouching the pin and this drived me crazy. When I change wires everything starts to work.

thank you all

closed 2022-07-16 by mauromorello