raomin/ESPAltherma #117: EAVH16S23DA6V - ERR: Time out on register ...

robertogarciaalvarez, 2022-06-24:
Dear all specially developers,

Thank you for this great work and for any help you could provide to me.

I have a Daikin Altherma with internal ( EAVH16S23DA6V) and external unit (EPGA14DAV3).
It is from 2021, so a fairly new machine.

I have followed this instructions using an ESP32 board. As described in instructions I have connected GPIO16 and 17 to X10A-pin2 and pin3 respectively. 

**Issue:**

I am not extracting any LabelDef values at all. I have then use the serial monitor at PlatformIO to understand what is going on. This is what I am getting:

------
Querying register 0x60... ERR: Time out on register 0x60! got 4/96 bytes
0x03 0x40 0x60 0x5c Querying register 0x10... ERR: Time out on register 0x10! got 4/16 bytes
0x03 0x40 0x10 0xac Retrying...
Querying register 0x10... ERR: Time out on register 0x10! got 4/16 bytes
0x03 0x40 0x10 0xac Retrying...
Querying register 0x10... ERR: Time out on register 0x10! got 4/16 bytes
0x03 0x40 0x10 0xac Retrying...
Querying register 0x10... ERR: Time out on register 0x10! got 4/16 bytes
0x03 0x40 0x10 0xac Querying register 0x20... ERR: Time out on register 0x20! got 4/32 bytes
0x03 0x40 0x20 0x9c Retrying...
Querying register 0x20... ERR: Time out on register 0x20! got 4/32 bytes
0x03 0x40 0x20 0x9c Retrying...
Querying register 0x20... ERR: Time out on register 0x20! got 4/32 bytes
0x03 0x40 0x20 0x9c Retrying...
Querying register 0x20... ERR: Time out on register 0x20! got 4/32 bytes
0x03 0x40 0x20 0x9c Querying register 0x21... ERR: Time out on register 0x21! got 4/33 bytes
0x03 0x40 0x21 0x9b Retrying..._
------

I have tried with several include/def/models and uncomment many LabelDef without any result in any of the attends.  
Any help would be much appreciated. 

Current seeting:

[image]
[image]

andreapier, 2022-06-24:
Connect the ground pin of x10a to a ground pin of your ESP32.

robertogarciaalvarez, 2022-07-15:
Thank you Andrea, I will try it out!

robertogarciaalvarez, 2022-07-17:
> Connect the ground pin of x10a to a ground pin of your ESP32.

Thank you @andreapier, after connecting the ground it is now sending data to MQTT broker. Fantastic! solved

raomin, 2022-07-17:
Glad you got it working @robertogarciaalvarez 
Thanks @andreapier, I guess it's still not big enough on the readme page... ;)

closed 2022-07-17 by raomin