raomin/ESPAltherma #118: m5stackplus show eo

jangarga, 2022-06-25:


jangarga, 2022-06-25:
Good morning,

after 2 months working fine, now m5stackplus show error in photo in attached and on home assistant data from altherma arrived not in continuos mode, attached screen. can you help me with this issue?

munzili, 2022-06-27:
Are you sure this is not a Network/Wifi issue?

I see some registers work. Does the Display show the same messages all the time (same messages when data correctly received from Home Assistant and also when not)?

raomin, 2022-06-27:
0x15 0xea means that the Heat pump did not understand the query from ESPAltherma.
It's either because the protocol of the heat pump is too old (not your case as it worked before) see eg #74  
Or, most probably here, a misconnection of a wire, probably ground... Make sure all your connections are clean and well pushed. 
You can also restart your heat pump but I doubt it will change much...

jangarga, 2022-08-04:
finally was network problem. thanks for support

closed 2022-08-04 by jangarga