raomin/ESPAltherma #120: COP calculation using a ShellyEM

alexdelli, 2022-07-08:
Hi everybody, anyone can help me with COP calculation using a shelly em?
I try this but COP is Zero every time for now. Is it possible to mix espaltherma sensors and Shelly sensors?

`      espaltherma_cop:
        friendly_name: "COP"
        value_template: "{% if is_state_attr('sensor.althermasensors','Operation Mode', 'Heating') and is_state_attr('sensor.althermasensors','Freeze Protection', 'OFF') %}
{{
((state_attr('sensor.althermasensors','Flow sensor (l/min)')| float * 0.06 * 1.16 * (state_attr('sensor.althermasensors','Leaving water temp. before BUH (R1T)') | float - state_attr('sensor.althermasensors','Inlet water temp.(R4T)')|float) )
/
(state_attr('sensor.shellyem_5e1757_channel_2_power') | float ))
|round(2)
}}
{% else %} 0 {%endif%}"
        unit_of_measurement: 'COP'`

thank you

BigFoot2020, 2022-07-09:
Use `states('sensor.shellyem_5e1757_channel_2_power')` instead of state_attr, the you should get actual power value from Shelly EM.

alexdelli, 2022-07-09:
> Use `states('sensor.shellyem_5e1757_channel_2_power')` instead of state_attr, the you should get actual power value from Shelly EM.

Thank you, I’ll try

alexdelli, 2022-07-09:
it works :)

closed 2022-07-09 by alexdelli