raomin/ESPAltherma #121: Broken JSON in case heat pump returns NULL value

DieterTHeck, 2022-07-13:
Hi,

due to whatever reason, my heatpumps returns "nothing" when querying 
{0x60,5,203,1,-1,"Betrieb_Störung_x60"}

This is taken over 1:1 in the MQTT JSON file, which causes "broken JSON format" errors downstream
[image]

I assume the code needs to be changed to put a null into the JSON in such cases

Thanks for investigating!

raomin, 2022-07-14:
Hi @DieterTHeck. I've seen where it is. Thanks for reporting. I'll fix it ASAP.

raomin, 2022-07-17:
Ok, it should be fixed now!

DieterTHeck, 2022-07-18:
Hi,
thanks, brilliant. 
I just 
1) updated converter.h with lastest version
2) recompiled without error
3) OTA update works fine with the buggy NULL value re-enabled
4) MQTT shows "-" now
[image]
5) Home Assistant templates work w/o errors now

Looks like it is working :-)

closed 2022-07-18 by raomin