raomin/ESPAltherma #123: No Data from Rotex RHBX11CB9W

Simon-ese, 2022-07-21:
I tried this tutorial with my Rotex RHBX11CB9W and was not successful. Has anybody had success with this heat Pump or one out of the family "ROTEX HPSU Niedertemperatur Bi bloc"?
Log:
```
ERROR: Wrong CRC on register 0x00!
Querying register 0x00... 
ERR: Time out on register 0x00! got 221/255 bytes
```
For now I have achieved to get information using the the Daikin LAN-Adapter(brp069a61) in combination with Node-Red. Unfortunately this only gives me very little useful Information:
- Indoortemperature, - Outdoortemperature, - OperationMode(cooling/heating), - TankTemerature, - TargetTemperature, - OperationMode(heating).
For my project I am technically only missing data about what the heat Pump is doing at the moment (heating Water / heating House).

raomin, 2022-07-21:
Check the ground connection from the ESP to the GND of the heat pump.

Simon-ese, 2022-07-22:
Thank you very much for your answer, I have checked the connection but it does not seem to be the problem. Connecting the esp with the 4 wires(5v,GND,TX,RX) to the heat pump does not work even though the esp is blinking. however connecting it to a external power source generates following MQTT Data on the LOG:
**-> I get this several Times**
```
ESPAltherma started!
Querying register 0x00... 
OK!
Adding registry 0x 0 to be queried.
Adding registry 0x10 to be queried.
Adding registry 0x20 to be queried.
Adding registry 0x21 to be queried.
Adding registry 0x60 to be queried.
Adding registry 0x61 to be queried.
Adding registry 0x62 to be queried.
Adding registry 0x64 to be queried.
```
**-> Every now and then I get:**
```
ESPAltherma started!
Querying register 0x00... 
ERROR: Wrong CRC on register 0x00!
OK!
```
**-> And sometimes I get::**
```
ERR: Time out on register 0x00! got 240/252 bytes
Querying register 0x00... 
ERROR: Wrong CRC on register 0x00!
�/h
ERR: Time out on register 0x00! got 164/172 bytes
0x7d 0x07 0xac 0x56 0x00 0x75 0x04 0x54 0x56 0x3a 0xdb 0x75 0x05 0x00 0x75 0x05 0x1e 0x9e 0xd5 0x05 0x00 0x75 0x05 0x00 0x75 0x05 0x07 0xaa 0xa9 0x00 0x75 0x05 0x00 0x75 0x05 0x00 0x75 0x05 0x2d 0x7f 0x07 0x00 0x00 0x35 0x00 0x00 0x75 0x05 0x00 0x75 0x05�
OK!
Querying register 0x00... 
ERR: Time out on register 0x00! got 125/125 bytes
0x2b 0xf4 0x7d 0x01 0x75 0x0a 0x83 0x9f 0x9f 0xb0 0x81 0x9d 0x01 0xfe 0xbe 0x00 0x00 0x9d 0xfe 0x00 0x75 0x05 0x9a 0x75 0x05 0xed 0x05 0x01 0x00 0x95 0x70 0x00 0x00 0x75 0x05 0x00 0xff 0x00 0x75 0x05 0x00 0x75 0x05 0x00 0x75 0x05 0x7e 0x7d 0xff 0x7c 0x7b}   �<�?�$�?�   0x05 0x47 0x07 0x75 0x05 0x0f 0x00 0xff 0x50 0x56 0x00 0x75 0x05 0x01 0x89 0x05 0x6f 0x75 0x09 0x75 0x05 0x00 0x75 0x05 0x00 0x75 0x05 0xe7 0xfe 0xee 0x0f 0xb2 0xff 0x0b 0x00 0x75 0x05 0x3f 0x00 0x75 0x18 0x75 0x05 0x00 0x75 0xf4 0x00 0xa1 0x67 0x00 0x00 0x75 0x05 0x01 0x3e 0x06 0x75 0x05 0x1c 0x56 0x56 0x00 0x75 0x05 0xc5 0xfc 0x7b 0xd0 0x01 0x07 0x14 
OK!
```
My heat pump:
> Daikin Altherma Gerät
> Lan Module: BRP069A61
> Modell : RHBX11CB9W
> Software Inside-Module: ID4562
> EEPROM Inside-Module: AS1706406-18A
> Software Outside-Module: ID0247
> Software Interface: v01.19.00
> EEPROM Interface: AS1705847-01F

What confuses me is that the tutorial is very straight forward. I am not sure if I made a mistake in the setup or if the problem is my heat pump. Is there a way to find out if my heat pump is compatible to ESPAltherma?

Thank you very much!

stomko11, 2022-07-22:
Can you post a photo of your wiring?

Simon-ese, 2022-07-25:
thank you for your answer.
[image]
[image]

mauromorello, 2022-09-13:
Hi; it was almost the same situation of mine, at the beginning;
It was caused by the faulty connection of dupont connectors; IE one wire is connected while the other no/blink

I suggest you to change the cable, or at least to tight with a very small screwdriver the female part of dupont connector on altherma side.

Sorry fow my english but I really want to help!

raomin, 2022-11-05:
@Simon-ese did you solve you issue?