raomin/ESPAltherma #124: Reverted switch

xalex75, 2022-08-09:
Hi.
I don't now why, but when the switch is off in HA, the relay is on, if I change the switch to on, relay goes off. I can't change it directly in the relay, how can I invert it in ESPAltherma?
Thanks.

raomin, 2022-08-09:
There are 2 types of relay:
- NC for normally closed (when off)
- NO for normally opened (when off)

You probably have a NC one. But have a look as some has 2 possible mode/connectors. 

I would advice against using NC, but as a failure would turn your heat pump always on... 

If you still want to invert the mode, just replace all HIGH to LOW and vice versa where you see 
`digitalWrite(PIN_THERM,`

xalex75, 2022-08-10:
It has a jumper, but if I move it, it doesn' switch anymore. So I think that changing it in code would be better, or change relay.
Thank you, I will try.

xalex75, 2022-08-10:
I bought new relay, I prefer not changing code too much, so fure update will be easier. It will arrive tomorrow, I hope that it will act correctly 😉

xalex75, 2022-08-11:
Hi. New different relays arrived, but these are active in inverted mode too.
When switch is on they are off and viceversa.
I think that I will create an inverted template switch in HA, if I'm the only one experiencing this.

raomin, 2022-08-14:
Hi @xalex75
Can you share your link to/picture of the module you bought.
Normally they all have 3 wire connector: one for the NC, one for the NO (and one for the COM)

DieterTHeck, 2022-08-14:
Turn it around and check. 
[image]

xalex75, 2022-08-15:
Hi. I think there's a misunderstanding. I know that I can use NC o NO contacts. What I mean is the tin my case when switch.alterma is off, the relay is active, when it's on, the relay is not active. Maybe it depends on the gpio used on Esp32 cam, I don't know, but I fixed that with a switch template in HA that has inverted state compared to switch.altherma.

[image]

There seems to be some inconsistency after reboots, but I have to check it better.

xalex75, 2022-08-21:
I'm still trying to use the switch in a reliable way.
Let's start from a situation with relay off.
When I reboot HA, the switch is in unknown state:
[image]

and the switch button in lovelace is shown as off.
If I click it, the entity state changes from unknown to off, and in lovelace it correctly shows it as off, but the relay goes on.
If I click it again, lovelace shows it as  on, and the relay goes off.

I created a template switch with inverted behavior compared to switch.altherma, and it would work if it was not for the unknown state after reboot, that makes the visible state in lovelace ureliable.

I'll try to get around this by testing the unknown state and activating automatically it one time to get the real state, it should work, but it look that there's some sort of malfunction in the switch part.

I notices thet in MQTT devices I see only althermasensor, not the switch. Is this correct?
Thanks.

xalex75, 2022-08-21:
Ok, it works, now switch.altherma state will not be unknown anymore.
It's still inverted, but I manage that inside home assistant.
I'm not sure if it keeps the current state across reboots, but I'll try to save it in a input_boolena and restore it after reboot.

`- alias: Riattiva Switch Altherma
  trigger:
  - platform: homeassistant
    event: start
  action:
  - delay: 00:00:30
  - condition: state
    entity_id: switch.altherma
    state: 'unknown'
  - service: switch.toggle
    entity_id: switch.altherma`

raomin, 2022-09-29:
So, all ok now @xalex75, I can close this?

xalex75, 2022-09-30:
Hi. Yes, I managed it in Home Assistant.
Thanks.

closed 2022-09-30 by raomin