raomin/ESPAltherma #130: M5StickCPlus and 2 relays for SG1 and SG2

dandjo, 2022-08-23:
# SmartGrid Features

Is it possible to control two relais to switch SG1 and SG2 with a M5StickCPlus? And if yes, could someone provide a circuit diagram? Thank you very much.

dandjo, 2022-08-23:
Can I use pin 32 and 33 of the M5StickC Plus? Which Relays do you recommend?

[image]

DieterTHeck, 2022-08-25:
Should work, check fotos here...
https://github.com/raomin/ESPAltherma/issues/17#issuecomment-1078870240

In the firmware there are 3 pins to be individually defined for SG1/SG2 and thermostat. 
Maybe try if they switch before you put it on your heat pump

raomin, 2022-09-29:
@dandjo can I close this one?

dandjo, 2022-09-29:
Yes, please close.

closed 2022-09-29 by raomin