raomin/ESPAltherma #131: compile error "invalid conversion from char to long int"

alxzndr, 2022-08-25:
When trying to compile with platformio i get this error message:

`file included from src/main.cpp:17:0:
include/comm.h:27:25: error: invalid conversion from 'char*' to 'long int' [-fpermissive]
   MySerial.print(prep, 4);
`

The build is working fine when you comment line 26 in incloude.comm.h 

ps: I dont know if it is related but i am building on a M1 macbook.

raomin, 2022-08-27:
That's very strange. 
The function is calling https://github.com/espressif/arduino-esp32/blob/adb01cf0c6c361a0823a4797bb08da3c2ea1b40d/cores/esp32/HardwareSerial.h#L118 which is to be called with `char *` like in the code.

Can you rightclick on `write` and `Go to definition` and paste it here?

You can also try to replace with this:
```c
  MySerial.write((uint8_t*) prep, 4);
```

alxzndr, 2022-09-01:
Thanks @raomin !

I replaced line 26 with:
`MySerial.write((uint8_t*) prep, 4);`

And now the build is working without any issues.

raomin, 2022-09-01:
OK I'll port it back to the code.

closed 2022-09-01 by raomin