raomin/ESPAltherma #133: ESPAltherma

Alexander-Wehner, 2022-08-31:
Hallo, ich wollte mal fragen ob ich das ESPAlthermaaucg bei meiner Daikin altherma verwenden kann?

Gruß Alex

[image]
[image]
[image]
[image]
[image]
[image]
[image]

raomin, 2022-08-31:
Hi @Alexander-Wehner 

I don't know for sure as it's a 13 year old model ; but it seems that there is a X10A port on your board, so it's definitely worth trying...
[image]

(please keep messages here in English)

Alexander-Wehner, 2022-08-31:
Hi,ok, what do i need?--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

raomin, 2022-08-31:
err.... you need to read the Readme document and follow the instructions...
https://github.com/raomin/ESPAltherma

closed 2022-09-02 by raomin

Alexander-Wehner, 2022-09-17:
Hello, I have an old Daikin Altherma heat pump, to this I have connected the M5StackC plus via the 8-pin X10A connector, I unfortunately do not get any data on my home assistant, what can be the reason? Greeting Alex

[image]

[image]

[image]

[image]

[image]