raomin/ESPAltherma #135: Error ets Jan  8 2013,rst cause:2, boot mode:(3,6) at NodeMCUV2

Goofman2, 2022-09-08:
Yesterday compiled program in platformIO for a NodeMCUV2 (ESP8266) and it worked. Now I cannot compile a working firmware.bin. Nothing chanegd as far as I can see. Don't know where to start....

raomin, 2022-09-08:
The error on the title is not a compilation error but a boot message from the ESP.
Give more details / screenshot of your issue.
(You can simply copy paste images here)

Goofman2, 2022-09-08:
I know, it's the serial debug port and as far as I cab see it's the WD.

[image]

Goofman2, 2022-09-09:
Can someone share complete .pio folder, maybe it's a version of the header files which are not correct.

raomin, 2022-09-29:
On ESP8266, some users reported issues with the use of PROGMEM.
Just remove it from the `default.h` definition file you're using and try again.

raomin, 2022-10-09:
Hello @Goofman2, what your situation now?

skrgahr, 2022-10-31:
i had same issue on NodeMCU v3. board did not start. with removing PROGMEM it finally started.

raomin, 2022-11-05:
I guess it's solved.

closed 2022-11-05 by raomin