raomin/ESPAltherma #138: Gas consumption

cybermerlo, 2022-09-09:
Hi guys!
I have an Altherma Hybrid with ESPAltherma since 2021.

I was wondering if it is possibile, in your opinion, to extract or calculate the consumption of gas (in cubic meter) using the data that ESPAltherma provide.
I would love to integrate that data into home assistant!

Thanks !

raomin, 2022-09-14:
I guess this might be possible.
I have clue on what information is available in a hybrid heat pump (mine is not) but I guess that you can track when the gas burner is ON.
I see some `Hot gas bypass valve (Y3S)` and `SV (hot Gas)` entries.
Assuming that the gas flow is constant, you can deduce a consumption in time then in m3... Home Assistant can do this.

Arnold-n, 2022-09-14:
For the hybrid system the gas flow is not constant, as the Intergas gas boiler modulates its heating output between 20% (user configurable, default even 30%) and 100% to reach a target leaving water temperature. Also, it uses gas both for heating water and for DHW (DHW is heated directly by the gas burner, not indirectly as in most other systems).

I think the hybrid reports the following information that can be used:
-leaving water temperature: temperature after gas burner
-water temperature of heating water before gas burner (after heat pump heat exchanger)
-valve settings that determine whether water flows through the gas burner or not
-whether the gas burner *should* be on (1)
-water flow of heating water

Based on these settings you can estimate how much heat is produced by the gas burner, and, based on an estimate for the burner efficiency, which depends on the water temperature before the gas burner (and the outside temperature), you can estimate how much gas is used for heating.

(1) I think that in a hybrid system the Daikin part only instructs the gas boiler to be on of off with a target leaving water temperature. If the gas boiler overshoots the target (which frequently happens due to the gas boiler's high minimum power setting, 20% of 36kW = 7kW, default 30%=11kW), the gas boiler goes into error mode, and in my system (2014) the Daikin controller ignores that; it does not try to down-regulate the heat pump setting to solve it, resulting in the system not reaching the desired target water temperature).

Estimating gas consumption for DHW is more difficult, as there are no temperature measurements before/after the gas boiler (only a target DHW) and there is no DHW flow measurement.

If you would have a smart gas meter (P1 output), and preferably no other gas consumers, you could combine its readings with the Daikin's status to get accurate gas consumption estimates both for DHW and for heating.

raomin, 2022-11-05:
Thanks @Arnold-n for the detailed answer, I'll close this issue.

closed 2022-11-05 by raomin

Arnold-n, 2022-11-06:
You're welcome @raomin . @cybermerlo: I learned recently that the newer hybrid systems do report information on gas consumption via P1/P2, even if they do not have an external gas meter (pulse signal) connected, and these systems do so separately for DHW and heating. We have yet to find out which units are used by these counters. Perhaps you can find these counters in your menu on the main controller too.