raomin/ESPAltherma #139: Custom board?

glosair, 2022-09-13:
Would anyone be interested in an opensource custom board If I design one? 

Things so far.

Switching power supply.
Serial voltage shifters. (5V to 3.3V).

andreapier, 2022-09-13:
I'd be interested, but only if the board also includes relays (ideally 4: 1x heating, 1x cooling, 2x Smart Grid). Only 2 for smart grid would still be cool, though.

glosair, 2022-09-13:
ESP32's aren't 5V tolerant, you might get away with it but.....

This board got extremely hot when connected as per the instructions.  When I installed level shifters and connected I also found my version K board would not power the ESP32 pictured via it's LDO regulator. 

[image]

[image]

mauromorello, 2022-09-13:
I used a LOLIN32 where there is a BMS onboard and can take from 3 to 7 volt in input

andreapier, 2022-09-14:
> No problems there, will have a look for suitable surface mount relays, will they need to be NO, NC or Both?

NO would work best. ESP Altherma supports both through a configuration switch, tough.

alexdelli, 2022-09-14:
with relays included yes

raomin, 2022-09-14:
Again, I don't want to oppose or anything. Building a board is a fun project, go for it!
But, also, consider having a look at [existing esp32 boards with relays](https://www.aliexpress.com/wholesale?SearchText=esp32+relay+board).
ESP32 are 5V (open) tolerant for IO pins, you should have no current dissipation (nor extra heat).
A good addition to these (existing? future?) boards would be proper connection to X10A. but then the cable length can greatly vary from installation to installation...

_I'll move this issue to the newly open discussion board_