raomin/ESPAltherma #140: Multi-Altherma monitoring into one platfrom

KindeG, 2022-09-14:
Dear Raomin,
First of all thanks for your great work, it's really something big!
My intention is to cover more heat pumps with MQTT query - tried to change the topic in mqtt.h line 3-4; but it ruins the integration into Home Assistant.
Do you maybe have a suggestion how to do it easily?

Thanks!

raomin, 2022-09-14:
Didn't think it would be a use case! ;)

You would need to dedicate topics per device. Basically renaming `espAltherma` to `espAltherma-1` (or anything else)
That would make something like this:

```c
#define MQTT_attr "espaltherma-1/ATTR"
#define MQTT_lwt "espaltherma-1/LWT"
```

And further in mqtt.h:
```c
      client.publish("homeassistant/sensor/espAltherma-1/config", "{\"name\":\"AlthermaSensors for Dev1\",\"stat_t\":\"~/STATESENS\",\"avty_t\":\"~/LWT\",\"pl_avail\":\"Online\",\"pl_not_avail\":\"Offline\",\"uniq_id\":\"espaltherma-1\",\"device\":{\"identifiers\":[\"ESPAltherma-1\"]}, \"~\":\"espaltherma-1\",\"json_attr_t\":\"~/ATTR\"}", true);
      client.publish(MQTT_lwt, "Online", true);
      client.publish("homeassistant/switch/espAltherma-1/config", "{\"name\":\"Altherma-1\",\"cmd_t\":\"~/POWER\",\"stat_t\":\"~/STATE\",\"pl_off\":\"OFF\",\"pl_on\":\"ON\",\"~\":\"espaltherma-1\"}", true);

```

This should work.

KindeG, 2022-09-14:
Well, you impressed me with the response time and solution - actually it seems to work like charm, let me just do the fine-tuning & testing and close the issue in few days :)!
Thanks Raomin!!

raomin, 2022-09-28:
All good in here? Can we close?

atomozero, 2022-09-28:
thank you, I need this possibility too. we have two heat pumps in the condominium and I didn't know how to read the data correctly

KindeG, 2022-09-29:
Hello Raomin,
yes, thank you very much, it definitely solves the issue. 
Have a nice one!

ralf2025, 2022-09-29:
Thank you very much, I also had this need and this post has solved it for me.

KindeG, 2022-10-11:
Hey Raomin,

one more thing: when doing multi-monitoring, the MQTT broker doesn't retain the messages - which causes quite a mess:

[image]

Like you only get the data when the actual query is happening. 
My code for sensors:
  - platform: template
    sensors:
      espaltherma_mario_dhw_setpoint:
        friendly_name: "DHW setpoint"
        value_template: "{{ state_attr('sensor.althermasensors_for_Mario','DHW setpoint') }}"
        unit_of_measurement: '°C'
      espaltherma_mario_lw_setpoint:
        friendly_name: "LW setpoint (main)"
        value_template: "{{ state_attr('sensor.althermasensors_for_Mario','LW setpoint (main)') }}"
        unit_of_measurement: '°C'

Maybe you have an idea how to set "retain" or get the last valid data for these sensors except "Unknown"?

Thanks!

raomin, 2022-10-11:
Retain is not really needed for the values, as HA is supposed to always be connected on MQTT.
But there might be something that explains what seems to happen: one is chasing the other one...
 
On `mqtt.h` line 70:
```c
    if (client.connect("ESPAltherma-dev", MQTT_USERNAME, MQTT_PASSWORD, MQTT_lwt, 0, true, "Offline"))
```
This first parameter `ESPAltherma-dev` is a client id. If a second client connects to the mqtt with the same client id, the MQTT server will assume that it's the same device and disconnect the first one...
So, you also need to change this value for the second device... sorry I missed it in the first place...

KindeG, 2022-10-13:
Thanks Raomin - fyi I've changed for:
 if (client.connect("ESPAltherma_<custom>-dev", MQTT_USERNAME, MQTT_PASSWORD, MQTT_lwt, 0, true, "Offline"))

now it works fine :)

closed 2022-10-13 by KindeG