raomin/ESPAltherma #145: ESPAltherma

Alexander-Wehner, 2022-09-17:
Hello, I have a 13 year old Daikin heat pump that I connected to the M5StackC Plus Stick with the 8 pin X10A connector. Which model should I select in the Setup.h file to get values on my mqtt?

[image]
[image]
[image]
[image]
[image]

raomin, 2022-09-19:
Try this one: `ALTHERMA(LT_CA_CB_11-16KW).h`

Alexander-Wehner, 2022-09-20:
[image]

[image]

[image]

[image]

raomin, 2022-09-20:
Check the logs, either on Home Assistant mqtt (see Readme) or with the serial monitor of PlataformIO while still connected to your esp32.

Alexander-Wehner, 2022-09-20:
[image]

[image]

raomin, 2022-09-20:
The timeout means that it's not getting anything from the heatpump.
Double check, triple check your connections, change your wire before losing hope! :)

Alexander-Wehner, 2022-09-20:
I have already checked connections several times and the cables have already been replaced, what else could be the reason?

mauromorello, 2022-09-20:
Sometimes these types of connectors are a little bit loose on altherma side and it's enough to mess all communication; try to change the wires;

Alexander-Wehner, 2022-09-24:
I got an ESP32 and installed the ESPAltherma code, but unfortunately I don't get a connection to the Home Assistant. What am I doing wrong??

Alexander-Wehner, 2022-09-29:
I'll check later, what if I still see it this way?

raomin, 2022-09-29:
When you did this screenshot, was the esp32 connected to the heatpump?

Alexander-Wehner, 2022-09-29:
no, that was still the M5StickC

Alexander-Wehner, 2022-09-29:
[image]

Alexander-Wehner, 2022-09-29:
[image]

Alexander-Wehner, 2022-09-29:
the screenshot is now with the ESP32

KalaNagHTD, 2022-09-30:
As a pre 2010 model your Daikin is too old most likely. It will also not work with the Daikin tool D-Checker which is only compatible with models from 2010 up.

raomin, 2022-09-30:
Many bizarre things... Why is it requesting register 0?
Choose a definition file, not the default.h
If it were a protocol issue, it would give 0x15 0x54...

Alexander-Wehner, 2022-09-30:
OK, too bad. Although I have the X10A slot--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

Alexander-Wehner, 2022-09-30:
ok, where can i find the definition file?--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

raomin, 2022-09-30:
Please read the Readme it's all described in there.

Alexander-Wehner, 2022-10-01:
I've now done everything again according to the instructions, unfortunately I still don't get any values.

[image]

[image]

[image]

[image]

raomin, 2022-10-04:
You have an ESP32 M5StickC but you selected the `env:esp32` which sets the communication ports to the heat pump to GPIO 16,17 instead of 36, 26. (see `setup.h`). Switch back to `env:m5stickplus`.

Alexander-Wehner, 2022-10-04:
no I have an ESP32--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

pitylee, 2022-10-09:
I have EKHBH(X)0016*** and struggling with the definition files and the lengths.

raomin, 2022-10-09:
@Alexander-Wehner , what you show us in the picture is an M5StickC Plus. 
[image]
If you're still using it, choose the right env.
If not, make sure your GPIO for the serial port are correct in the `setup.h`

@pitylee can you clarify your problem, and create another issue unless you have the same behaviour?

Alexander-Wehner, 2022-10-10:
Hallo, dies ist mein aktuelles setup, aber ich bekomme keine Werte

[image]

[image]

Sandrohd75, 2022-10-13:
Hello Alexander, 
I have also an "old" EKH BX 0008 AA3V3 from 2009, so similar to you. I connected M5StickC Plus, in the meantime I get some bytes / parts of the value back. My problems could be the not perfect connection on the X10A connector, I will now ordering a better connector. When I am looking on your picture of the connector with the 8 Pins, it loks you are using the wrong pining. I used the same as the 6pin connector in the picture in the attachment, so PIN 1, 2,3  and 5 as GND. As mentioned I get some values. 
Eventuell nochmal auf Deutsch, schau Dir die Belegung des Steckers nochmal an, Du verwendest Pin 1, 3, 4 und 8. Meines erachtens ist das der Grund warum Du keine Verbindung bekommst. Wo hast Du die Belegung her? Prüfe mal meine Belegung und gebe nochmal bescheid, ich hoffe das mein Stecker am Wochenende geliefert wird, kann dann nochmal Rückmeldung geben, ob ich dann die Daten nun vollständig auslesen kann. 
Ciao Dirk aus Heidelberg
 
[image]

Alexander-Wehner, 2022-10-13:
Hallo Sandrohd75

Ich bin nach dieser PIN Belegung gegangen, welche PINs sollte ich mal versuchen?
Gerne auch über PN

Sandrohd75, 2022-10-13:
Hallo Alexander, leider habe ich nur ein unscharfes Bild aber es geht auch ohne Bild. 
Wir in den alten Daikins haben den X10 Stecker mit 8 Pins, die neuereren Daikins haben 5 Pins, hatte in der nachricht zuvor 6 Pins geschrieben, es sind aber nur 5.
Du musst an Deinem Stecker die gleiche Pinbelegung verwenden wie bei den neuen Daikins, da die Pins sehr klein sind bitte genau nachschauen und zählen
Habe Dein Bild von oben verwendet und die korrekte Belegung in Blau ergänzt, hoffe es ist lesbar:)
[image]

Alexander-Wehner, 2022-10-14:
[image]

raomin, 2022-11-05:
Please keep this discussion in English so everyone can benefit from it.
@Alexander-Wehner did you eventually solve your issue?

Sandrohd75, 2022-11-06:
Dear Alexander, sorry for delayed feedback. In the meantime i get the perfect connector, so it works now, 
I find out, the your connector pining is correct, so I deleted the wrong information from some days ago. 
[image]

My connector:
[image]

I used the updated/newest release of ESPAltherma with the "Basic implementation of Rotex/Daikin S protocol". After some testing with the system and updating the StickC-Plus with the newest software I get the values over MQTT in my node-red. 
Screenshot of my values, not su much but better than nothing:
[image]

I hope with this informations you will be able to hack your EKHB, but I can confirm that a EKH BX 0008 AA3V3 from 2009 runs with ESPAltherma.
Thanks also to Raomin and all people who worked on this project.

Alexander-Wehner, 2022-11-06:
Hello Sandrohd75, sounds interesting, where did you get the plug from? (Link) and where did you get the firmware for the Stick C Plus?
Gladly also via PM
greeting

Sandrohd75, 2022-11-06:
plug at Ama..... ,see Screenshot, 
[image]
And update of the software for the stick with the M5-Burner-software, but perhaps this has no influence? 
[image]

Alexander-Wehner, 2022-11-06:
Thanks. Cable is ordered, how accurate is your PIN assignment now?
1: GND
3: TX
4:RX
8:5V

Sandrohd75, 2022-11-06:
1: 5V
3: TX
4:RX
8:GND

nicolasvila, 2022-11-21:
> 

Hi @Sandrohd75 , are you sure of the pin out? (for RX & TX)  this is not what's on ESPAltherma page:
https://raomin.github.io/ESPAltherma/doc/images/rotexX10A.png
[image]
I have the same heat pump and was not sucessful to get values from registers... Only got time out so far...

Sandrohd75, 2022-11-21:
Hi Nicolasvila, I can confirm this pinout works. Please double check your wires and the connector, this also caused some troubles on my side. Perhaps in the meantime Alexander-Wehner tested also with the proposed wires and connectors.
@Alexander-Wehner: Did it works on your side now?

I am still working on the values and perhaps getting more values/data. Here some graphs from today.

[image]

nicolasvila, 2022-11-22:
> Hi Nicolasvila, I can confirm this pinout works. Please double check your wires and the connector, this also caused some troubles on my side. Perhaps in the meantime Alexander-Wehner tested also [...]

Hi. Indeed you were right. I had several issues at the same time. Wrong board selected in my platformio.ini,  defect on the cable connection and wrong profile for my heat pump which is quite old (2009 I guess)
Like some users, I'm using an Altherma HP with a 8-pins X10 connector so I made custom 8-pins Dupont connectors in order to avoid problems.
I need to stick on PROTOCOL_S_ROTEX or PROTOCOL_S (except 0x50 registers) HP profiles
Now everything is fine and I'm receiving values from the HP. Looking good.
I'm glad you posted a screenshot of your homeassistant as I'm a bit puzzled by the temperature values I'm receiving from the HP: I was expecting the "outdoor air temperature" to be close from the real outdoor air temperature... Values seem way too high, even on your installation. What do you think?

nicolasvila, 2022-11-22:
Here are some pictures of my connections on the HP and my ESP32 (AZ-Delivery, pins 16 and 17) Does anyone know where is located the +5V on a 8-pins X10 connector? I didn't want to take a risk, so I'm using a USB power source for now.
[image]
[image]
[image]
I put below a link to the ESP32 pinout diagram:
https://cdn.shopify.com/s/files/1/1509/1638/files/ESP_-_32_NodeMCU_Developmentboard_Pinout_Diagram.jpg

Sandrohd75, 2022-11-22:
That looks good! With the values of outdoor I am also not perfectly shure, but today, nobody at home, so less heating I get also simlar values like you show.

[image]

andrejb2, 2022-11-24:
I put the power of esp32 on pin 1 (the first pin on the x10a connector). the last pin 8 is gnd.

what value you have for outside temperature?
{0x54, 4, 153, 2, -1, "Outlet Water Temp.(C)"},
{0x54, 4, 103, 2, -1, "Outdoor air temp.(C)"},

as for this definition this one is my Outlet Water Temp and not outdoor air temp.

i have a EKHBH016BA9WN indoor unit

nicolasvila, 2022-11-24:
I have an EKHBH008 unit and I still have the wrong labels from PROTOCOL_S.h 

- Indoor Suction Air Temp.(C)            21,22 (indoor temperature measured by the HP?)
- Indoor Heat Exchanger Temp.(C)    24,02  ("cold" water coming back to the HP?)

The two other values are very close and one of them should be the hot water at the exit of the HP
I think values are not related with air temperature because they're too high. Water temperature make more sense
- Outdoor air temp.(C)                       29,95  
- Outdoor heat exchanger temp.(C)   28,78

Alexander-Wehner, 2022-11-26:
[image]

Hello, I had time today to try the new plug and what can I say, I get values. Thanks for the tip. Maybe someone can help me visualize the values in Home Assistant?

andrejb2, 2022-11-26:
just follow https://github.com/raomin/ESPAltherma#integrating-with-home-assitant

it is really simple to integrate

Alexander-Wehner, 2022-12-03:
@nicolasvila hi, have you been able to read out more values? I would still be interested in the outside temperature and the flow rate.

nicolasvila, 2022-12-04:
Hi @Alexander-Wehner , I have only  been able to read values according to the PROTOCOL_S_ROTEX definition, despite the fact my heat pump is a "Daikin Altherma" and there is no reference to Rotex anywhere
For now that's fine but I suppose more values could be retrieved (for instance all the variables from PROTOCOL_S return something but mapping seems to be wrong or meaningless)  I would also like to read outside temperature. 
I haven't tested yet the application DChecker to see if more values are readable but it's in my TODO list.  I tried to unzip the app and to check configuration files but there's no plain text files, only binary files  (.ldd if I'm corect)

Alexander-Wehner, 2022-12-04:
hi, thank you maybe you will get the value of power consumption too

Alexander-Wehner, 2023-03-02:
@nicolasvila, hast du schon Zeit gefunden mehr Werte auslesen zu können? 

[image]

[image]

Mich würden, die Außentemperatur, COP Wert noch Interesieren 

So sieht es bei mir aus.

nicolasvila, 2023-04-20:
@Alexander-Wehner No I didn't to try to read more values than the one you are using. My Altherma unit is quite old and I guess it has less sensors than more recent ones