raomin/ESPAltherma #146: Finding all registries

kaskii, 2022-09-18:
I found "list registries.txt" file in the repos doc folder and I am wondering how has it been generated? I'm interested to get a list of all possible information I could get from my pump and was hoping there is already a way to dump whole registries.

raomin, 2022-09-18:
Hi @kaskii.

You could dump all the bytes in the pages for your device but there is no way to know what it corresponds to and how to convert it to meaningful values.
The required registry information mainly comes from Daikin diagnostic tools.
So, all the information that this tool queries and displays for diagnostic is there.
So, if you get a matching definition file for your heat pump, you will have in it all the information that can be displayed.

kaskii, 2022-09-19:
Thanks for pointing me into right direction! 

I found definition files inside a tool and with a little decompiling said tool I managed to write a nice console application to easily convert definition files (.ldd) straight to ESPAltherma supported format.

The tool I found was pretty old version and it seems Daikins update URL is giving timed out errors currently so updating the tool doesn't work. Maybe the URL has been changed. I'll try to get more up to date files.

Now I just need to figure out how to change some values. 👀

closed 2022-09-19 by kaskii