raomin/ESPAltherma #148: ERPA14 & ETVH Unit

pro2call-nl, 2022-09-21:
Hello Is it possible to add the ERPA14 and ETVH units?

I would like to see the R9T and R10T sensors so i can calculate COP. 

[https://my.daikin.eu/dam/document-library/Installer-reference-guide/heat/air-to-water-heat-pump-high-temperature/EPRA014-018(DV_DW)_EPRA14-18(DV7_DW7)_ETVH(X)16E(6V_9W)(7)_Installer%20reference%20guide_4PEN644737-1A_English.pdf]

Page 260 and 261.

what can i do to get the correct data?
[Outdoorunit schematic.pdf](https://github.com/raomin/ESPAltherma/files/9620102/Outdoorunit.schematic.pdf)

pro2call-nl, 2022-10-02:
> your

I have found the correct temperatures. 

I have also asked my installer to offer me an EKPCCAB4 with the correct software. so we can extract data.

raomin, 2022-11-05:
I guess we can close this one. Let me know if you get an fresher D-Checker tool...

closed 2022-11-05 by raomin