raomin/ESPAltherma #153: double Thermostat ON/OFF value in MQTT with different content

HWL1968, 2022-09-30:
Dear reader,

I am using ESPAltherma to get "all" the values / measurements of a Daikin Hybride heatpump.

System: Daikin Altherma hybride EHYHBX08AV3
Monitoring: M5StickC

Issue: in both the 'espaltherma/OneATTR/Thermostat' topic as 'espaltherma/ATTR' the attribute 'Thermostat ON/OFF' is being used twice for different registers (0x10 and 0x60). In my system one of them keeps ON (register 0x60; while the heatpump is OFF) and the other switches according to the state of the heatpump (0x10). I assume that both registers have a different meaning, since it became an issue after switching the heatpump on since it's getting cold indoor.

proposed solution: change the name of the topic (for example thermostat1 and thermostat2; this way it can be checked what the meaning of ...1 and ...2 is).

Best regards.

DieterTHeck, 2022-10-28:
Note that you can edit the names yourself in the code. In the preconfigured config files you had to select in the beginning. For instance to Thermostat_x60 or so.

raomin, 2022-11-05:
I do agree that one would need to be renamed.
Did you find what do they correspond to? If not, I'll go with ...1 and ...2

HWL1968, 2022-11-06:
Sorry for the late respons. Yes, I did find the place to change the names, but I opt to keep my version of ESPAltherma consistent with git (and since I am no experienced developer I don't know how to edit it in Git). Therefor I wanted to make a report of this 'issue'. 

@raomin: sorry, I still couldn't figure this out. Our thermostat has been connected to X2M 1 and 4 as an ON/OFF switch (we are using EvoHome with their new heatpump ON/OFF module). What I notice is:
- it seems like switching ON by the EvoHome module switches on the thermostat2 (x60 register)
- thermostat2 switches OFF with a delay (makes sense since switching to fast isn't that good of an idea)
- but it stays ON even after the system is stopped completely and can be ON until the next EvoHome action which triggers the heatpump to start
therefor: no clue so far. But it is related to the ON/OFF of the thermostat. Maybe some delay setting of the system? I couldn't find anything yet however.

THK56, 2023-02-20:
My understanding for the two thermostats: the one in Reg 0x10 is for the outdoor unit, the other in Reg 0x60 is for the indoor unit. I observed that the IU thermostat only has an impact, when the heating is controlled based on the room temperature (and not based on LWT, what is mostly the case). When the room temperature raises over the set point plus 1/2 of the configured hysteresis, the IU thermostat changes to OFF, as long the room temperature is higher as set point minus 1/2 hysteresis. Then it switches to ON again (like a simple to 2-point controller. Usually the HP controls room temp within hysteresis (they call it "modulation"), but when the sun additionally heats the room the temp can raise up over the control band.
The function of the OU thermostat is not clear to me. Sometimes it's inverted to the IU thermostat...