raomin/ESPAltherma #156: COP calculation with primary/secondary current

dandjo, 2022-10-01:
Hi Altherma Fans.

So far I have calculated the capacity of the compressor and the COP using "INV Primary Current" and the voltage of my grid. However, this results in large fluctuations and apparently time-shifted data. If I divide the current of the pulsed 3-phase output signal "INV Secondary Current" by 3 and multiply it by the voltage "Voltage (N-phase)", I get much more plausible values for the power of the compressor and ultimately for the COP. Additionally I consider the power factor at lower currents. I packed this data into a function based on some measurement data. Here is the result of the compressor power calculation (openHAB ECMAScript).

```
var current = parseFloat(itemRegistry.getItem('espaltherma_compressor_secondary_current').state);
var voltage = parseFloat(itemRegistry.getItem('espaltherma_compressor_voltage_n_phase').state);
var emergency = itemRegistry.getItem('espaltherma_emergency_active').state;

current = current / 3;  // 3-phase sum

var power = 0;

if (current > 0 && voltage > 0 && emergency === OFF) {
  var power_factor = 0.752385 * Math.pow(current, 0.11962) - 0.00141164;
  power_factor = power_factor > 1 ? 1 : power_factor;
  
  power = current * voltage * power_factor;
  power = Math.round(power * 100) / 100;
}

events.sendCommand('espaltherma_compressor_secondary_power', power);
```

The heating power is calculated as follows (openHAB ECMAScript).

```
var power = 0;

var operation_mode = itemRegistry.getItem('espaltherma_operation_mode').state.toString();
var defrost_operation = itemRegistry.getItem('espaltherma_defrost_operaton').state;

if (operation_mode === 'Heating' && defrost_operation === OFF) {
  var flow = parseFloat(itemRegistry.getItem('espaltherma_flow_sensor').state);
  var lw_temp = parseFloat(itemRegistry.getItem('espaltherma_leaving_water_temp_before_buh').state);
  var iw_temp = parseFloat(itemRegistry.getItem('espaltherma_inlet_water_temp').state);

  var diff_temp = lw_temp - iw_temp;
  
  var water_density = 0.993;  // for approx 35°C

  var convert_flow = 60 * water_density; // converts l/min to kg/h for water
  var thermal_capacity = 1.1626; // of water in Wh/(kg•K)

  power = flow * convert_flow * thermal_capacity * diff_temp;
  power = Math.round(power * 100) / 100;
}

events.sendCommand('espaltherma_heating_power', power);
```

These are my graphs of the measured data (Grafana from InfluxDB).

[image]
[image]

What do you think? Would appreciate a discussion about it.

sahni2021, 2022-10-05:
I also noticed large fluctuations calculating the COP using the primary current, but never thought of any for the secondary current. Now when I use the calculation below (inside Home Assistant) I agree with @dandjo in that it delivers more plausible values.

`{{
((state_attr('sensor.althermasensors','Flow sensor (l/min)')| float * 0.06 * 1.16 * (state_attr('sensor.althermasensors','Outlet Water Heat Exch. Temp. (R1T)') | float - state_attr('sensor.althermasensors','Inlet water temp.(R4T)')|float) )
    /
  (state_attr('sensor.althermasensors','INV secondary current (A)') | float / 3 * state_attr('sensor.althermasensors','Voltage (V)')|float / 1000))
  |round(2)
}}`

dandjo, 2022-10-05:
What I have seen so far: primary current always precedes the heat generated, i.e. the compressor starts up, produces heat and shortly afterwards this results in higher temperatures. That's not the case with Secondary Current, which makes it implausible again. Also, dividing the current by 3 may not always result in correct values, especially at higher values.

sahni2021, 2022-10-06:
I think to solve the problem with the current ambiguity the only solution might be to use an external device (e.g. Shelly 3EM) to measure the currents in real-time. Something I rather avoid due the high costs for the Shelly device and the cable clutter in the circuit braker box ;-)

Kepro, 2022-10-13:
I think the issue is with `Voltage (N-phase) (V)` it's not stable from 75 it goes up to 320 and my unit is on 1 phase so there should be around 230-240V always...

dandjo, 2022-10-13:
Hi @Kepro be aware that N-Phase is the current after the SSR as I understood, not the current of your grid. It's the current after the transformation for the compressor motor itself. It's a PWM signal.

Kepro, 2022-10-13:
hi @dandjo thanks for the quick reply, I just found a couple of formulas for COP, so I used it and many people complain about the same from my understanding... so is this then correct? https://github.com/raomin/ESPAltherma#calculating-cop

Kepro, 2022-10-15:
@dandjo one more question, it's possible to use `Voltage (N-phase) (V)` or `INV secondary current (A)` or `INV primary current (A)` to calculate current load? I have 1 phase external unit 6kW

dandjo, 2022-10-15:
`Voltage (N-phase) (V)` is the voltage **after** the "transformer" for the compressor. `INV secondary current (A)` is the current **after** the "transformer" for each phase. We assume that all 3 phases have the same voltage (I'm not sure if this is the case).

The other option is to mutiply `INV primary current (A)` with the voltage of your grid, since this is the current **before** the "transformer".

I'm not quite sure which method is better or more accurate in the end.

stskrp, 2022-10-19:
Tried both options (primary and secondary currents), but getting too big COP:
[image]

pwwukpw, 2022-10-28:
Sorry new here .. 

I'm getting 

INV primary current (A): 0
INV secondary current (A): 0
Voltage (N-phase) (V): 203

Though COP appears to be calculated

[image]

Would be great to get accurate COP

pwwukpw, 2022-10-29:
Managed to grab a capture with Amps and Volts  showing ..  V way above Grid voltage and both primary and secondary current. Installer did put a grid meter just before the heat pump but I think it's dumb. Any consensus as to which formula to use for accurate COP. With so many variables I'm keen to optimise operation for best lowest consumption. 

[image]

raomin, 2022-11-05:
I'll move this to the discussion section, and I would welcome any suggestion to improve the COP calculation on the readme :)