raomin/ESPAltherma #158: EHVH04S18EJ6V unit

Matssa56, 2022-10-02:
Hi!

So after getting my ESP8266 to work, I can't seem to find the correct definition files. Here are those that I tested:
  - default,
  - ALTHERMA(BIZONE_CB_04-08KW).h (tried it since my PAC is 4Kw)
  - ALTHERMA(LT_DA_04-08KW).h
  - ALTHERMA(HYBRID).h
  - ALTHERMA(LT_CA_CB_04-08KW).h
  - ALTHERMA(MONOBLOC_CA_05-07KW).h

My PAC has the X10a with 5 pins.

Here is the message that I receive most of the time (I uncommented everything in the def files to be sure that I see everything to test):

```
 "*Refrigerant type": "Conv 801 not avail.",
  "Override CMD:0x00 N_INV=1": "Conv 996 not avail.",
  "Override CMD:0x00 N_STD=0": "Conv 996 not avail.",
  "Override CMD:0x00 N_FAN=2": "Conv 996 not avail.",
  "Override CMD:0x00 N_EV=2": "Conv 996 not avail.",
  "Override CMD:0x00 N_20S=1": "Conv 996 not avail.",
  "Override CMD:0x00 N_CH=1": "Conv 996 not avail.",
  "Override CMD:0x00 N_SV=3": "Conv 996 not avail.",
  "Sensor Data Qty": 0,
  "INV compressor Qty": 0,
  "STD compressor Qty": 0,
  "Fan Data Qty": 0,
  "Expansion Valve Data Qty": 0,
  "4 Way Valve Data Qty": 0,
  "Crank Case Heater Qty": 0,
  "Solenoid valve Qty": 0,
  "Max. connectable indoor units": 0,
  "Connected Indoor Unit Qty": 0,
  "O/U MPU ID (xx)": 0,
  "O/U MPU ID (yy)": 0,
  "O/U capacity (kW)": 0,
  "NextDataGrid": "Conv 995 not avail.",
  "In-Out separator": "Conv 998 not avail.",
  "Override All Clear": "Conv 996 not avail.",
  "WifiRSSI": "-75dBm"
```

On almost all values I get 
```
Querying Registery 0xXX
TimeOut! Check connection
```

My wish is to have the power consumption in Wh or kWh, pump/compressor status and the different temperature values. I have the documentation and all but doesn't really seem to help.
My guess is that this PAC is too recent (got it in Feburary) :S

Thanks a bunch!

raomin, 2022-11-05:
Coming back late on this one, sorry.
If you have a _TimeOut! Check connection_ message it is not an issue of file definition but a issue with your... connections.
Check the wires and connections. 
Make sure your GND is connected to the heat pump and your ESP.

Matssa56, 2022-11-07:
> Coming back late on this one, sorry. If you have a _TimeOut! Check connection_ message it is not an issue of file definition but a issue with your... connections. Check the wires and connections. Make sure your GND is connected to the heat pump and your ESP.

Hey,

Thanks for the feedback, I just checked and I have 5V between the red and black wire, so my guess is that my GND and VCC are good. Other proof is that my esp turns on ^^

However I didn't check the TX and RX, not sure how I can check that.

Anyways, I'm gonna buy some connectors so that my installation is better since I used several Dupont wires in order to make an extension...
I'm also gonna try and ESP32 in case my esp8266 doesn't work, however do you know if I can at the same time use a USB cord with my PC and also have the VCC/GND linked to the Altherna ? I'm afraid to make a mistake in terms of voltage.
Also, if the esp doesn't find any wifi, can it still work if I plug it in my USB and launch the upload and debug stuff?

raomin, 2022-11-19:
Sorry for the late reply. JST connectors will not work. The 2 plastic bars they have on the side get in the way...

raomin, 2022-12-20:
Hey @Matssa56, did you solve your issue?

Matssa56, 2022-12-21:
> Hey @Matssa56, did you solve your issue?

Hi,
Not yet I only received the pins yesterday so I'll have to build them and see if the connection is better. Gonna also try to move my wifi setup in order to use my ESP32 instead of my esp8226.

Matssa56, 2022-12-23:
Hey again, so it was indeed a connection issue, my Dupont wires must have had issues, using a connector works fine.
Next stop : change the default stuff to get the correct values and make the HA screen.
[image]

closed 2022-12-23 by Matssa56

Matssa56, 2022-12-23:
> Hey @Matssa56, did you solve your issue?

Hey quick question, how can I manage to find some missing values? I tried two header files and I can't seem to find my water pressure (1.8 bar). Should I try maybe adding some registry keys to the default one?

raomin, 2022-12-23:
Some HP do not return water pressure. Look in the closed issues, it was discussed.