raomin/ESPAltherma #160: ESPAltherma queries drawback to previous version - starts to read according to a different setup

KindeG, 2022-10-05:
Dear Raomin,

we are using ESP32 NODEMCU boards for reading ESPAltherma, and the situation is like this:
I'm setting up a query for a lot of parameters, which reads really nice - for a couple of circles. But after something happens, it draws back to a previous version - with really less parameters reading, even on a more frequent data pull.

I couldn't trace back where the thing is - but for sure after sending "OK": 

reconnect();
  mqttSerial.println("OK!");

  initRegistries();
  mqttSerial.print("ESPAltherma started!");

Do you maybe have an idea? initRegistries() calls what?

logs:
Message 342 received on espaltherma_DACEHU1/log at 11:15:
Done. Waiting 5 sec...
QoS: 0 - Retain: false
Message 341 received on espaltherma_DACEHU1/log at 11:15:
Querying register 0x61... 
QoS: 0 - Retain: false
Message 340 received on espaltherma_DACEHU1/log at 11:15:
Querying register 0x60... 
QoS: 0 - Retain: false
Message 339 received on espaltherma_DACEHU1/log at 11:15:
ESPAltherma started!
QoS: 0 - Retain: false
Message 338 received on espaltherma_DACEHU1/log at 11:15:
Adding registry 0x61 to be queried.
QoS: 0 - Retain: false
Message 337 received on espaltherma_DACEHU1/log at 11:15:
Adding registry 0x60 to be queried.
QoS: 0 - Retain: false
Message 336 received on espaltherma_DACEHU1/log at 11:15:

QoS: 0 - Retain: false
Message 335 received on espaltherma_DACEHU1/log at 11:15:
OK!
QoS: 0 - Retain: false
Message 334 received on espaltherma_DACEHU1/log at 11:14:
Done. Waiting 15 sec...
QoS: 0 - Retain: false
Message 333 received on espaltherma_DACEHU1/log at 11:14:
Querying register 0x64... 
QoS: 0 - Retain: false
Message 332 received on espaltherma_DACEHU1/log at 11:14:
Querying register 0x62... 
QoS: 0 - Retain: false
Message 331 received on espaltherma_DACEHU1/log at 11:14:
Querying register 0x61... 
QoS: 0 - Retain: false
Message 330 received on espaltherma_DACEHU1/log at 11:14:
Querying register 0x60... 
QoS: 0 - Retain: false
Message 329 received on espaltherma_DACEHU1/log at 11:14:
Querying register 0x30... 
QoS: 0 - Retain: false
Message 328 received on espaltherma_DACEHU1/log at 11:14:
Querying register 0x20... 
QoS: 0 - Retain: false
Message 327 received on espaltherma_DACEHU1/log at 11:14:
Querying register 0x10... 
QoS: 0 - Retain: false

raomin, 2022-10-09:
I fear I don't really understand your issue. But I see that ESPAltherma is rebooting at message 339:
```
Message 339 received on espaltherma_DACEHU1/log at 11:15:
ESPAltherma started!
```
If this happens often enough, can you log it on Monitor (USB) so we might get more info...?

KindeG, 2022-10-11:
Well, actually it's more an issue for ESP32 board. After deleting flash and re-writing it worked really fine. 
Thanks for the attention!