raomin/ESPAltherma #165: ERROR: Wong CRC on register 0x60!Quering register 0x61...

leonidnf, 2022-10-13:
Hi, community!
first of all, thank you @raomin, and contributors for the project! It's lovely to have such a tool for controlling and monitoring our heating/cooling device!

I have a slight problem with connecting my M5stickplus to the HPSU and I hope that I'll get a little help.

I'm getting the following error  on the M5stickplus LCD:
ERROR: Wong CRC on register 0x60!Querying register 0x61...

Here is my setup:
HPSU: EHSXB08P50DA3

Uncommented `HPSU6_ULTRA` and no custom code added to the project.
Already checked the ground cable and connected it to the metal case of HPSU. When the ground cable is removed, I'm getting another error.
[image]
[image]
[image]
[image]

TomOudeNijhuis, 2022-10-16:
I had the same issue and switched back to an older version. After some experimenting I think that pull request #149 broke things. 

Have tried to find the exact source of the issue, but no luck yet. Must say that mixing the protocols in `comm.h` has not made the code easier to read. There is some mixed usage of `unsigned char`, `char` and `uint8_t` as well, but fixing this did not resolve the issue.

raomin, 2022-10-16:
Thanks @TomOudeNijhuis for the feedback. I'll have a deeper look and report here.

agrenott, 2022-10-16:
Hi @TomOudeNijhuis , could you please un-comment the logBuffer(buffer, len); on line 118 of comm.h (right before the get CRC) and post the resulting logs?

raomin, 2022-10-16:
From the picture, looks like it's for 0x60 and 0x61 
I haven't tested #149, I will tomorrow, but my understanding is that `unsigned char` or `char` shouldn't change anything.

raomin, 2022-10-16:
Actually, you're not the only one reporting some CRC errors while still having some values properly reported.
This intermittent issues could either be caused by a faulty cabling or, if this is fixed by reverting to a previous commit before #149, then it might be the change introduced by the `Serial.flush()`.
For this one, @agrenott was right to add the parameter to false as it is intended to make sure we do not have ghost values on the input before sending our queries, so I don't think it's this.

Now if @TomOudeNijhuis reverted before #128, this could be because of the removal of the wait time between registries queries... [(here](https://github.com/raomin/ESPAltherma/commit/840378a49357c3da59c10d116af61399a01484b1#diff-34d21af3c614ea3cee120df276c9c4ae95053830d7f1d3deaf009a4625409ad2L257)
I really need to do some more testing....

andreapier, 2022-10-17:
This is happening to me too. Can confirm this only happens with registers `0x60` and `0x61`. Other registers work fine.
I double checked my wiring just in case, but it looks to be just fine.
Let me know if you need more info, thanks.

agrenott, 2022-10-17:
I guess this is indeed linked to the unsigned vs. signed mix-up, as the equality isn't verified for values > 127:
https://onlinegdb.com/S3x5e2t9_
[image]

I can't reproduce, but I'll submit a PR.

leonidnf, 2022-10-17:
Hi @raomin, @agrenott, @TomOudeNijhuis,
I've just tested the `Fix CRC computation` PR on my system and the error still persists.
Also, I've checked out a commit before #149 and it works without errors.

Please let me know if I can help you with something

agrenott, 2022-10-17:
Hi @leonidnf , sad my change doesn't fix the issue.
Can you please uncomment the logBuffer line and attach the failing case?
https://github.com/raomin/ESPAltherma/blob/main/include/comm.h#L118

You can fetch it from the MQTT log topic, or the serial console if you connect to it with USB cable.

agrenott, 2022-10-17:
I've just pushed a 2nd commit with a bug fix which is so obvious I can't understand how only 0x60+ register would be impacted, as I basically truncated all protocol I replies by 2 bytes...
@leonidnf if you can test it please.

leonidnf, 2022-10-17:
@agrenott Thank you for the fast reply and for solving the issue!
I'm happy to confirm that the bug is in the past! Now I see clean logs

raomin, 2022-10-17:
Seen the PR, looks good now, thanks @agrenott! I'm merging.

PR #167 (2022-10-17): Fix CRC computation
- Align unsigned char usage, to ensure proper comparison results.

Fix #165.
commit 4e42875f (2022-10-17): Fix protocol I reply length decoding

Fix #165
commit 84248b03 (2022-10-17): Fix CRC computation

Align unsigned char usage, to ensure proper comparison
results.
Fix #165.
closed 2022-10-17 by raomin