raomin/ESPAltherma #166: Timeout for Older Altherma (EKHBH 11-16kwh) on 0x53 0x55 0x54 got xx/18 bytes

pitylee, 2022-10-13:
I have added a new definition file that [can be found on my fork](https://github.com/pitylee/ESPAltherma/commit/10b51629ebfcbe9e93bb8f7e477152e1fee89b29) but getting these errors:

```
ERR: Time out on register 0x53! got 15/18 bytes
...
ERR: Time out on register 0x53! got 16/18 bytes
...
ERR: Time out on register 0x55! got 15/18 bytes
...
ERR: Time out on register 0x55! got 16/18 bytes
0xff 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
...
ERR: Time out on register 0x55! got 17/18 bytes
...
ERR: Time out on register 0x54! got 17/18 bytes
...
ERR: Time out on register 0x54! got 15/18 bytes
...
ERR: Time out on register 0x54! got 16/18 bytes
...
ERR: Time out on register 0x54! got 14/18 bytes
...
ERR: Time out on register 0x53! got 14/18 bytes
...
ERR: Time out on register 0x53! got 17/18 bytes
...
```

Detailed data can be found in the attached data.txt.

[data.txt](https://github.com/raomin/ESPAltherma/files/9780986/data.txt)

agrenott, 2022-10-17:
The output you're getting looks like garbage; I can't even see the register ID (0x53/0x54/0x55) anywhere, while it should prefix any reply. You're also getting variable reply length to the same query, which shouldn't happen.

I'd say something's probably wrong on the physical layer.

pitylee, 2022-10-17:
How?

raomin, 2022-10-17:
Are you sure you have common GND between the altherma and the esp32? Can you recheck your cable and provide pictures of the setup?

raomin, 2022-11-05:
@pitylee Any update on this one? Did you get your connection ok?

pitylee, 2022-11-05:
Hello. Unfortunately not yet. I have ordered new cables, but I think the
best would be to have the original socket, which I am unable to find.

pitylee, 2022-11-08:
I just did it  <3 

This is the one: 
https://www.conexelectronic.ro/ro/conectori-nslnsg-254396mm/17132-CONECTOR-CU-CABLU-8-PINI-MAMA-2-54MM-30-CM.html

andrejb2, 2022-11-18:
i just ordered an esp32 and put espaltherma on it.
I have a 2010 version of EKHBH016BA9WN
using this definition
`LabelDef PROGMEM labelDefs[] = {
    // 0x50 not supported by EKHBH016BA9WN (no data here for the two values pulled)
    {0x50, 0, 103, 2, -1, "HP Sensor(kgcm2)"},
    {0x50, 2, 103, 2, -1, "LP Sensor(kgcm2)"},

    // 0x53 seems to be a collection of booleans
    {0x53, 0, 152, 2, -1, "EV (pls)"},
    {0x53, 0, 200, 1, -1, "Circulation pump"},
    {0x53, 1, 164, 1, -1, "Andrej"},
    {0x53, 2, 164, 1, -1, "Outdoor Fan (Upper)(rps)"},
    {0x53, 3, 164, 1, -1, "Outdoor Fan (Lower)(rps)"},
    {0x53, 3, 200, 1, -1, "External heater?"},
    {0x53, 4, 200, 1, -1, "INV Comp. Frequency(Hz)"},
    {0x53, 5, 200, 1, -1, "Priority to domestic water"},
    {0x53, 6, 200, 1, -1, "Burner inhibit from solaris"},
    {0x53, 8, 200, 1, -1, "20R (SV) Output"},
    {0x53, 7, 200, 1, -1, "20S (4-way) Output"},
    {0x53, 10, 200, 1, -1, "Crankcase Heater"},
    {0x53, 11, 200, 1, -1, "Ener-Cut Output"},

    // Testing stuff....
    {0x54, 0, 153, 2, -1, "Refrig. Temp. liquid side(C)"},
    {0x54, 1, 103, 2, -1, "Andrej1"},
    {0x54, 2, 153, 2, -1, "Inlet water temp.(C)"},
    {0x54, 3, 153, 2, -1, "Andrej3"},
    {0x54, 4, 153, 2, -1, "Outlet Water Temp.(C)"},
    {0x54, 6, 153, 2, -1, "D(C)"},
    {0x54, 7, 153, 2, -1, "Andrej7"},
    {0x54, 8, 153, 2, -1, "DHW tank temp.(C)"},
    {0x54, 10, 103, 2, -1, "F(C)"},
    {0x54, 8, 109, 2, -1, "Discharge pipe temp.(C)"},
    // Byte 12 seems to be a 1 byte value, evolving during day in continuous way
    // 0x1b mid-day, 0x24 mid night -> some delta temp??
    {0x54, 12, 101, 1, -1, "Water heating temp(deg)"},
    {0x54, 13, 151, 1, -1, "R/C Setpoint(C)"},
    {0x54, 14, 153, 1, -1, "andrej14"},
    {0x54, 15, 153, 1, -1, "andrej15"},
    {0x54, 16, 153, 1, -1, "andrej16"},
    {0x54, 16, 151, 1, -1, "andrej17"},
    {0x54, 16, 109, 1, -1, "andrej18"},
    {0x54, 19, 153, 1, -1, "andrej19"},
    {0x54, 20, 153, 1, -1, "andrej20"},
    {0x54, 21, 153, 1, -1, "andrej21"},
    {0x54, 22, 153, 1, -1, "andrej22"},

    {0x55, 0, 201, 1, -1, "Operation Mode"},
    {0x55, 1, 204, 1, -1, "Error Code"},
    {0x55, 2, 204, 1, -1, "Thermo Off Error"},
    {0x55, 3, 204, 1, -1, "Warning Code"},
    {0x55, 4, 204, 1, -1, "Caution Code"},
    {0x55, 5, 151, 1, -1, "andrej55x5"},
    {0x55, 6, 204, 1, -1, "andrej55x6"},
    {0x55, 7, 204, 1, -1, "andrej55x7"},
    {0x55, 8, 204, 1, -1, "andrej55x8"},
    {0x55, 9, 204, 1, -1, "andrej55x9"},
    // TODO: seems to reply on 0x56 as well, but not used by dchecker...
    {0x56, 0, 103, 2, -1, "????"},
    {0x56, 1, 103, 2, -1, "????1"},
    {0x56, 2, 103, 2, -1, "????2"},
    {0x56, 3, 103, 2, -1, "????3"},
    {0x56, 4, 103, 2, -1, "????4"},
    {0x56, 5, 103, 2, -1, "????5"},
    {0x56, 6, 103, 2, -1, "????6"},
    {0x56, 7, 103, 2, -1, "????7"},
    {0x56, 8, 103, 2, -1, "????8"},
};

// Override protocol
#define PROTOCOL 'S'
`
a lot of testing attributes, just comment or delete them

I do not understant the third value (103, 152, 200, 304, 151, 153,...); did not see it anywhere described in the code.

andrejb2, 2022-11-18:
and response
`{
    "EV (pls)": 256,
    "Circulation pump": "ON",
    "Andrej": 0,
    "Outdoor Fan (Upper)(rps)": 0,
    "Outdoor Fan (Lower)(rps)": 0,
    "External heater?": "OFF",
    "INV Comp. Frequency(Hz)": "OFF",
    "Priority to domestic water": "OFF",
    "Burner inhibit from solaris": "OFF",
    "20R (SV) Output": "OFF",
    "20S (4-way) Output": "OFF",
    "Crankcase Heater": "OFF",
    "Ener-Cut Output": "OFF",
    "Refrig. Temp. liquid side(C)": 19.5938,
    "Andrej1": 48.0742,
    "Inlet water temp.(C)": 24.1875,
    "Andrej3": 4.09375,
    "Outlet Water Temp.(C)": 24.0156,
    "Outdoor air temp.(C)": 24.0156,
    "D(C)": 24.0938,
    "Andrej7": 132.094,
    "DHW tank temp.(C)": 42.5156,
    "F(C)": 0.00390625,
    "Discharge pipe temp.(C)": 85.0312,
    "Water heating temp(deg)": 25,
    "R/C Setpoint(C)": 0,
    "andrej14": 0,
    "andrej15": 0,
    "andrej16": 0.640625,
    "andrej17": 164,
    "andrej18": 1.28125,
    "andrej19": 0,
    "andrej20": 0,
    "andrej21": 0,
    "andrej22": 0,
    "Operation Mode": "Heating",
    "Error Code": " 0",
    "Thermo Off Error": " 0",
    "Warning Code": " 0",
    "Caution Code": " 0",
    "andrej55x5": 16,
    "andrej55x6": " 0",
    "andrej55x7": " 0",
    "andrej55x8": " 0",
    "andrej55x9": " 0",
    "????": 48.0039,
    "????1": 120.188,
    "????2": 0.46875,
    "????3": 0,
    "????4": 0,
    "????5": 0,
    "????6": 0,
    "????7": 0,
    "????8": 0,
    "WifiRSSI": "-81dBm",
    "FreeMem": "227248"
}`

raomin, 2022-11-19:
The third indicates which `converter` to use to transform the bytes into value. See `converters.h`

Sandrohd75, 2023-02-11:
Hello andrejb2, 
did you find in the meantime some additional "values" or "register" which could be useful for an Daikin Altherma version from 2010. I am still testing some of the unclear values. I will keep you updatet...