raomin/ESPAltherma #173: M5 Stick C Plus dead after

haz-69, 2022-10-20:
First time evere using PlatformIO sp i think i do something wrong.
Opened the folder ESPAltherma-main. and then entered my settings for WIFI and MQTT in src/setup,h. Nothing else where changed.
After pressing F1  : PlatformIO: Upload them M5 Stick C plus is totaly dead?
What have i been missing?

haz-69, 2022-10-20:
After F1 PlatformOP:Upload PlatforIO says: enviroment esp32 SUCCESS  but it is still dead?

irecord, 2022-10-20:
Does the M5 Stick power up if you plug the USB cable in to power it? Can you confirm if you are getting 5v from the connector?

haz-69, 2022-10-20:
Yes when it is programmed with Easyloadr_M5STickC_Plus_FactoryTest, but not after above
[image]

irecord, 2022-10-20:
What did you do to solve it?

haz-69, 2022-10-20:
Changed settings in platformIO rx and tx pin eas not defined and wrong environment

closed 2022-10-20 by raomin