raomin/ESPAltherma #176: How to enable/disable the on/off thermometer...

trantoriana, 2022-10-24:
Hi,

First of all thanks for the excellent piece of firmware and a very clear description on how to use it.

Unfortunately I am puzzled how to enable/disable the on/off thermostat relay from outside (Step 3 (optional)).
Where for step 4 - grid control a MQTT is used, I cannot find such a thing for the on/off thermometer.

Am I just blind or is the method how to set/unset the relay missing from the (otherwise excellent) documentation.

Thanks, Trant

(Also a further question on connecting the relay. I see in the picture it seems connecting pin 30 & 35 on X2M. Unfortunately I cannot find any description about this in the installation documentation I have gotten... How did you know to use these two connections?)

trantoriana, 2022-10-24:
Trant read further down....

"switch.altherma activates the relay connected to the PIN_THERM"

Clearly a case of rtfm.

Stays the question about x2m.30 and x2m.35. How do I know which to connect....

DieterTHeck, 2022-10-28:
Check the installation manual of your Daikin device. In my case the thermostat and also SG and the likes were pretty well described in the section for external devices.

trantoriana, 2022-10-29:
Hi Dieter,

Thanks for the suggestion, however I am not a electrician so I have trouble interpreting the info contained in the optional equipment booklet.

I base my assumption the 30&35 can just be connected through the ON/OFF relay based on following in the booklet:
[image]

However, on the same page there also is:
[image]

and this is confusing.. I have no idea what the actual products displayed are or whether they just utilize the connections in a particular product, based on the boxed items.

In both pictures the connections 30 & 35 seem to be involved in the main LWT zone.

My assumption is that I just need to connect 30 & 35 to turn on the heating and disconnect to turn off the heating (there by simulating a ON/OFF thermostat...

Right?

raomin, 2022-10-29:
Yes 30+35 will turn heat on.
(30 with 34 will turn colling on)

DieterTHeck, 2022-10-29:
I have another device, but it looks like the second one. It works just like roamin said. 

Don’t touch 3/4 from the upper picture. Looks like 230V.

trantoriana, 2022-10-29:
Hi Raomin, Dieter,

I will put the relay in the box connected to 30&35 with thicker wires, and leave my esp32 outside.

I scrolled back to the main page and the schematic picture connecting to the relay is the same as my second picture...
In hindsight it is al so simple :-)

Thanks for the support.

Trant.

closed 2022-11-05 by raomin

tkroon89, 2023-01-02:
> Trant read further down....
> 
> "switch.altherma activates the relay connected to the PIN_THERM"
> 
> Clearly a case of rtfm.
> 
> Stays the question about x2m.30 and x2m.35. How do I know which to connect....

I have the same question, but how to do this via MQTT? I don’t use Home Assistant, and want to control the relay via MQTT. Thank you for the support!

raomin, 2023-01-02:
Hi @tkroon89,
You can interact with the switch through it's command topic: espaltherma/POWER
Send it `ON` or `OFF`

Cheers.

tkroon89, 2023-01-03:
Works like a charm now. Thank you Raomin!