raomin/ESPAltherma #181: Thermostat on/off in Leaving water temperature mode

Badlee2020, 2022-10-31:
Hi all, 

First of all, it is awesome what has been put together here!

I have a Rotex hpu hybrid (RHYHBH08AAV3) with a BRP069A61 lan module. All in all the system does not really run correctly. I am hoping that with this or perhaps the collected general knowledge someone can help me. 

A few things about my system. It is used in a 2 family house, i have two hybrid cubes, the room heating uses leaving water temperature mode (room temperature mode does not provide enough heat, the controller is in the heating room and the Temperature is all over the place, it could be that i have not found the right settings). On my system the lan module gives me the SG1/2 contacts that are connected to the solar inverter. The intent was to use the extra solar energy to produce warm water. Unfortunately, if i am lucky, the heat produces warm water once a week using the smart grid function ( also in summer when the room heating is off). There is no reason as to when I runs. 

From what I understand the thermo on/off only works using the room thermostat to set the leaving water temperature? Has anyone had success getting the BRP069A61 to work for producing warm water using the espaltherma as controller? If the x85a was also connected to espaltherma could that assist in forcing the system to produce warm water? 

Any and all suggestions would be greatly appreciated!!!

DieterTHeck, 2022-10-31:
Hi, I have a totally different setup, but the thermostat just enables room heating here. When thermostat = off, it's like the room heating being in summer mode. The actual heat pump still runs on auto pilot with thermostat enabled or disabled (i.e. It heats the water tank whenever the water in there gets too cold). I think what you want /need is smartgrid not thermostat? From your text above, the solar inverter commands are not understood by SG? Are you sure they are connected properly / compatible? The relevant on/off combinations for SG1/2 are shown in the wiki.

DieterTHeck, 2022-10-31:
Ah, and you need to enable SG and thermostat in the Rotex settings menu. It may ignore it otherwise!

Badlee2020, 2022-10-31:
thanks for the quick answer!!

Yes, I am looking at getting the smart-grid function to work, my guess is also that the signal is not being interpreted correctly.  I am also looking at controlling the room heating a bit better (in terms of turning it off according to the weather forcast), but that is not the most important thing. 

After all the trouble I have had with the system in trying to use the smart-grid function, anything is possible. 

I am more than willing to look at everything again and test anything. 

The BRP069A61 module (https://www.daikin.eu/content/dam/document-library/Installer-reference-guide/ctrl/BRP069A61,%20BRP069A62_4PEN464229-1_2017_08_Installer%20reference%20guide_English.pdf) is a bit different than what is described in ESpaltherma. I think that the principle should be the same.

The BRP069A61 has a SG0 and SG1, that are very similarly configured as the SG1/SG2. 

[image]

From my understanding the SG0/1 are simple contact circuits that when closed are "turned on".  At least that is what the installer described it as. From the way it looks, using the Solar Inverter’s output switch when a specific amount of power is produced the Smart-grid tuns on/off. Also, when simply shorting the contacts connecting the SG0 contact 1 to the SG0 contact 2 the light also turns on and off. Who knows, there might be more logic / digital signal behind it.  Recently I have connected the ESPaltherma to the sg0/1 contact to relays controlled by ESPaltherma. The SG light turns on and off when SG commands are sent to the ESPaltherma.

My Rotex only has a "LAN module" installed setting that needs to be enabled and 4 other settings that so the communication with the Lan modul and the control is given to the Lan module. Those are set and i think it works otherwise the once a week usage of solar power would not be possible :)  

thanks in advance!!

Badlee2020, 2022-11-04:
thanks again for the help. 

the issue was fixed and the ESP including SG contacts work like a charm. The issue was that when Daikin installed and configured the Lan Module they changed some setting that made the heater prefer gas over the heat pump. After having a closer look at all the settings, the issue was found.

closed 2022-11-04 by Badlee2020