raomin/ESPAltherma #183: Disconnect before WiFi

EvertJob, 2022-11-02:
Hey guys,

I'm using the M5StickC PLUS ESP32 and it works really well execpt one issue: After an update it won't connect to the WiFI.
The display shows: `Restoring previous sate: OffSetting up wifi... Connecting to Joep`

There is only one work-around: Disconnect 5V and GND, turn off the device, turn it back on, wait till it's connected to WiFi and connect is back with 5V and GND.

Is this normal behaviour?

alexdelli, 2022-11-02:
Did you try to re-reflash the firmware via USB?

EvertJob, 2022-11-02:
Yes. Same issue.

raomin, 2022-11-05:
This is very weird. I cannot recall having the same issue.
We did some changes in the way we handle wifi disconnection, but I don't think it could be linked.
Can you open a serial monitor and display what you get after reflashing via USB?

EvertJob, 2022-11-06:
Will check! The most strange thing is that after disconnecting the ground and VIN the problem is gone, could connect to WiFi and plug it back in.

raomin, 2022-11-06:
Maybe some noise in the signal from the gnd/5v of the heatpump that would weaken the WiFi signal reception...? What is the reported rssi on altherma power vs fully disconnected on battery?

OnktheMonk, 2022-11-27:
I have the same issue. But it happens every half hour. Disconnecting and reconnecting solves the issue but it keeps returning.

EvertJob, 2022-11-27:
Did you turn on OTA updates? I’ve disabled OTA and upload the new code with an USB cable and all the problems are gone.

OnktheMonk, 2022-11-27:
No i didn't turn this on. But I just started 2 days ago so i have the latest version.

OnktheMonk, 2022-11-29:
Maybe bad connection with the dupont connectors on the PCB. I now ordered a correct connector. I'll see if this will improve.

alexdelli, 2022-12-01:
> Where can I find the right connector for the pcb? Orded 2 already and they
> are both wrong.

https://m.it.aliexpress.com/item/1005003388263149.html?spm=a2g0n.order_detail.order_detail_item.3.6e8b582fmxa4Ey&gatewayAdapt=glo2ita

EvertJob, 2022-12-02:
When disabling OTA it's working fine.

closed 2022-12-02 by EvertJob