raomin/ESPAltherma #187: Daikin Altherma 3C Gas D2CND

mehmetu61, 2022-11-13:
Hello, is there a possibility that your product will work with this model?

I tried it but unfortunately it didn't work

[image]

raomin, 2022-11-19:
I don't know this model, it looks like there is no X10A port. I see RX/TX/5V on X10 and X9.
You could give it a try but I would not be very optimistic....

mihsu81, 2024-03-20:
X9 is for the DRGATEWAYAA. I've tried the X10 but I only get the below errors in the logs continuously.
```
Retrying...

Querying register 0x10... 
ERR: Time out on register 0x10! got 2/12 bytes
0x80 0x00 

Querying register 0x20... 
ERR: Time out on register 0x20! got 2/12 bytes
0x80 0x00 

Querying register 0x21... 
ERR: Time out on register 0x21! got 2/12 bytes
0x80 0x00 

Querying register 0x30... 
ERR: Time out on register 0x30! got 2/12 bytes
0x80 0x00 

Querying register 0x60... 
ERR: Time out on register 0x60! got 2/12 bytes
0x80 0x00 

Querying register 0x61.. 
ERR: Time out on register 0x61! got 2/12 bytes
0x80 0x00 
```

Any chance to understand which protocol it is?

raomin, 2026-10-23:
It's been a while! The Altherma 3C Gas is a condensing boiler with another protocol on its port, not an Altherma heat pump, so not something ESPAltherma can read. Closing with the 2.0 release.

closed 2026-10-23 by raomin