raomin/ESPAltherma #191: WiFi connection fall down

coralblack12, 2022-11-18:
I'm using a module NodeMCU 8266, I'm able to see Altherma data for days but suddenly the card loose connection and the red light start blinking very fast. Even if I try to reset the card or disconnect the power, nothing gonna change. I've replaced the card with a new one, but after one week, the problem is the same.
Only if I reprogram the card with same configuration it starts connecting again but just for few days. What's could be the problem?
Thanks

raomin, 2022-11-19:
Do you see logs on Serial/USB when the card is blinking red?

raomin, 2022-11-24:
I too witness connection loss, might have been introduced by a recent code change.
It looks like this in the graphs, and the AlthermaSensor entity becomes _Unavailable_
[image]

I'll keep an eye on it.

coralblack12, 2022-11-24:
At the moment I have assigned a fixed IP to the Altherma in order to exclude DHCP problem. Since now no issue, but let's wait a couple of weeks

raomin, 2022-11-24:
I've set an alert on my phone in case of loss of connection. 
On my side it happened during heating phase.

coralblack12, 2022-12-04:
Even if I've set the static IP address it seems the problem is still occuring. Now my suspect is about the power supply. My ESP is using the power of Altherma pin X10A.
I've tried to disconnect that connection and connecting a USB power supply but, doing this, the card is stopping to providing data to home assistant anymore.

raomin, 2023-01-02:
Looks like there are issues with the wifi selection when multiple AP broadcast the same SSID. In this case, the ESP might randomly choose the weaker one. See #224 and #213 
Is it your case?

coralblack12, 2023-01-03:
Dear Raomin, 
It seems I have solved in this way:
- replacing the ESP32 with a different model. Now I'm using the narrow one, with pin RX2/TX2 instead of GPIO 16 and 17.
- connecting the power via USB cable and still connecting the ground to the Altherma pin.

In the last month I did not noticed any WiFi interruption.
Thanks

closed 2023-03-12 by raomin