raomin/ESPAltherma #194: Home assistant sensors not available

OnktheMonk, 2022-11-26:
I cannot get the sensor info available in HA I see the attributes and added the example code in configuration.yaml 
I can add them in cards but they display no info. Also did a restart.

tarmor1, 2022-11-28:
Maybe could you share some example from configuration.yaml and print screen on how they display in cards?

OnktheMonk, 2022-11-29:
```
# Configure a default setup of Home Assistant (frontend, api, etc)
default_config:

# Text to speech
tts:
  - platform: google_translate

group: !include groups.yaml
automation: !include automations.yaml
script: !include scripts.yaml
scene: !include scenes.yaml

cover:
  - platform: group
    name: Screens
    entities:
      - cover.badkamer_2
      - cover.bureau_2
      - cover.kinderkamer_2
      - cover.slaapkamer_2
      - cover.woonkamer_raam_2
      - cover.woonkamer_schuifraam_2

camera:
  - platform: mjpeg
    name: OctoPrint
    still_image_url: http://192.168.88.238/webcam/?action=snapshot
    mjpeg_url: http://192.168.88.238/webcam/?action=stream

notify:
  - name: hass agent Desktop Hendrick
    platform: hass_agent_notifier
    resource: http://192.168.88.30:5115/notify
  - name: "Mail Riccardo"
    platform: smtp
    server: smtp.gmail.com
    port: 587
    timeout: 15
    sender: xxx@gmail.com
    encryption: starttls
    username: xxx@gmail.com
    password: !secret gmailHendrick
    recipient:
      - yyy@hotmail.com
    sender_name: "Hendrick"

sensor:
  - platform: template
    sensors:
      espaltherma_operation:
        friendly_name: "Operation mode"
        value_template: ""
      espaltherma_iuoperation:
        friendly_name: "Indoor Operation mode"
        value_template: ""
      espaltherma_dhw:
        friendly_name: "DHW Temp"
        value_template: ""
        unit_of_measurement: '°C'

#ads:
#  device: 5.32.252.106.1.1
#  port: 851
```

[image]

If you need any other info. Please let me know.

tarmor1, 2022-11-29:
Hi!
You defenitely need `value_template` to be populated. https://www.home-assistant.io/integrations/template#value_template
This is how I have the DHW tank temp (it's in HA terms "legacy template"), but works still. Anyway, you definitely need to have it set.
```
      dwh_water:
        friendly_name: "DHW tank temp. (R5T)"
        unit_of_measurement: '°C'
        value_template: "{{ state_attr('sensor.althermasensors', 'DHW tank temp. (R5T)') }}"

```
Try your value template out in Home Assistant Developer Tools > Templates:
[image]
.. and you can observe the actual value (43C in right side taken from the Attribute set by espaltherma)

OnktheMonk, 2022-11-29:
That was the issue! Works now. Thank you for the info!

AbeltjeNL, 2022-12-06:
@OnktheMonk, how did you manage to get the energy info into HA? Which unit model do you have and what config file do you use? I can't seem to read this info from my Altherma.

OnktheMonk, 2022-12-06:
Yeah I get the info. But I have some issue with connectivity, I guess bad Dupont connectors. I orded a correct connector for the pcb and I'm waiting for them to arrive. Then I'll continue this project. Now I lose connection every half hour.

I have the EHVX08S23EJ6V.

AbeltjeNL, 2022-12-06:
> Yeah I get the info. But I have some issue with connectivity, I guess bad Dupont connectors. I orded a correct connector for the pcb and I'm waiting for them to arrive. Then I'll continue this project. Now I lose connection every half hour.
> 
> 
> 
> I have the EHVX08S23EJ6V. 

Alright, but which /include/dev/file.h do you use?

OnktheMonk, 2022-12-07:
I used this one: #include "def/ALTHERMA(LT_DA_04-08KW).h"

AbeltjeNL, 2022-12-07:
> I used this one: #include "def/ALTHERMA(LT_DA_04-08KW).h"

Alright, thank you. Can you tell me how you managed to read the or make the energy sensors in HA? I can't seem to wrap my head around it. Are you using the COP calculation?

OnktheMonk, 2022-12-07:
Not yet but I parked the project because of the connection loses until I get the right connectors. 

Here you can find a very good explanation (in Dutch) to get the info. But I guess that wont be an issue. 

https://gathering.tweakers.net/forum/list_message/73682572#73682572

AbeltjeNL, 2022-12-07:
Dankjewel! 😁

closed 2023-01-02 by raomin

swalsh11, 2023-02-19:
I have a similar Issue,  all of the data comes through in MQTT, 
[image]
however  when I set: 

```
      dwh_water:
        friendly_name: "DHW tank temp. (R5T)"
        unit_of_measurement: '°C'
        value_template: "{{ state_attr('sensor.althermasensors', 'DHW temperature(R5T)') }}"
        

```
it returns ""None"
[image]

This happens with all the entities, any help appreciated.  
        `

tarmor1, 2023-02-20:
Hi!

> ```
>         value_template: "{{ state_attr('sensor.althermasensors', 'DHW temperature(R5T)') }}"
> ```
Are you sure your MQTT main element is `sensor.althermasensors`?
You get None if either attribute or main element is misspelled...

And btw, I recommend moving to the non-deprecated notion of sensors. This is how I have it. Using this you get as well HA Statistics and some other nifty stuff, eg getting icons according to device classes.
```
template:
  - sensor:
      - name: "LW setpoint (main)"
        unit_of_measurement: '°C'
        device_class: temperature
        state_class: measurement
        state: >
          {{ state_attr('sensor.althermasensors', 'LW setpoint (main)') }}

```

(And the binary states eg Booster heater, BUH heaters etc define as `binary_sensor` under `template`):
```
  - binary_sensor:
      - name: "Defrost Operation"
        device_class: power
        state: >
          {{ state_attr('sensor.althermasensors', 'Defrost Operation') }}
```

swalsh11, 2023-02-20:
Thanks, I has the name incorrect somehow it was :
 sensor.altalthermaermasensors 
[image]

This Code Works:

```
template:
  - sensor:
      - name: "LW setpoint (main)"
        unit_of_measurement: '°C'
        device_class: temperature
        state_class: measurement
        state: >
          {{ state_attr('sensor.altalthermaermasensors', 'LW setpoint (main)') }}
          

```

swalsh11, 2023-02-20:
I have updated the sensor names and shared them here, Thanks for the help:

https://github.com/swalsh11/ESPAltherma-Template