raomin/ESPAltherma #196: Desinfektion ALTHERMA(LT_DA_04-08KW).h

EvertJob, 2022-12-02:
Hey guys,

Is there an register to get the disinfection state of the heatpump?

raomin, 2022-12-20:
Hi EvertJob! 
I'm not using this (I was told it was useless, at least in France).
I cannot see what could relate to this in the [list of values](https://github.com/raomin/ESPAltherma/wiki/Information-about-Values)...

bluemoon-duck, 2023-02-23:
Hi EvertJob, are you possibly referring to the DHW legionnaires disease prevention by heating the DHW tank to at least 60 deg on a regular interval (once per week)? If so, I believe that's rather a schedule you have to set than it is a considered an operational state of the HP. But I'm new to the subject (have my HP for two weeks now).

habitoti, 2023-08-24:
I created a binary „disinfection“ sensor that goes to true if DHW setpoint is 60 degrees…which obviously only works properly if this is not also your comfort setpoint 😉

luksa, 2024-10-06:
In my case `{0x62,2,305,1,-1,"Storage comfort ON/OFF"}` goes to `ON` when disinfection is active. And that's the only time it's ON.