raomin/ESPAltherma #203: WIFI connects only when Heat pump switched off

AjubaGit, 2022-12-07:
First of all - Yeah, it works - Thanks a lot for this great development.

Only a minor issue: Wifi connects only when the heat pump is switched off and I start the m5StickC with the internal accumulator. Then it connects quickly to my Wifi and I can start the heat pump. 
Basically Wifi works fine, even in cellar where Wifi is weak. Only problem, once it loses connection it does not reconnect as long as heat pump is on. 
m5StickC  is power supplied from the heat pump.
Of course I could place a repeater to increase Wifi strength but I do not know if this solves the basic probloem of not connecting to my Wifi while heat pump is on.

Any solution recommended?

And just a question about security: In the definition file I have to write the Wifi password and MQTT password in clear text. Any possibility to handle this safer?

Thank you for hints.
Andi

raomin, 2022-12-08:
Hi @AjubaGit, welcome.

Have a look at the wifi rssi value reported in the sensor values. Less than -70 (eg -80) is very weak. You can also increase the `FREQUENCY` [here](https://github.com/raomin/ESPAltherma/blob/8879bcc4e9db38e700c35c98b7d55bb96fb3a07f/src/setup.h#L18) so you have less traffic, less opportunity to disconnect.
 
It's common that when the signal is weak, the heat pump operation creates noise and weakens the signal too much to establish proper connection.
What you could do is increase the distance from the heat pump with longer cable (a cat5 network cable should be good for that), hence increasing the distance from the noise (and closer to the wifi AP at the same time if possible)

Regarding security the credentials in the setup.h, it stays where you put them, in a file on your drive... If this is a concern for you, just empty it when you're done flashing, and clear your compiled code.

AjubaGit, 2022-12-09:
Hi Raomin

Thank you for the quick and professional support. I have set a repeater close to the heat pump and now it connects without issues. Wifi is now at -43.

Thank you not only for the great project but also for the detailed fool proof documentation which encouraged me to start.

Enjoy your beers.
Andi

seb-koch, 2023-12-07:
I have the same issue, except my M5StickC only connects when i disconnect it from the Heatpump. I assume the HP connection somewhat interferes with the sticks built in antenna. Thoughts?