raomin/ESPAltherma #204: [Question] Altherma 3 RF+W with WT32-ETH01

pstarkl, 2022-12-08:
Hi everyone!

I´m trying to use esp-altherma with my heat-pump (EHBX08EF9W) with a WT32-ETH01
I have successfully swapped the WiFi-Connection with an Ethernet-Connection and deploying the project was also successfuly.

However when I connect the WT32 to my heat-pump, my payload only looks like this:
[image]

But I enabled a lot more values than one, as seen here:
[image]

I also hooked up the WT32 to the heat-pump and used the serial monitor, which gave me this output:
[image]

So I assume, the reading from 0x00 was successful after 3 attempts, but those multiple timeouts are not normal I guess.
What do I have to change in my settings in order to bring this project to work properly? (Baud-rate etc?)

Thanks a lot

raomin, 2022-12-08:
Check your connection. A complete time out means that ESPAltherma didn't get anything from the heat pump.
Pay close attention to your GND connection, it should be properly linking ESP GND to the Heat pump GND.
Sometimes, dupont wires are broken with no sign outside.

pstarkl, 2022-12-10:
Hi, thanks for your reply, I really appreciate it.
I already triple checked the connections, im using PIN4 for RX and PIN5 for TX on the WT32, according to this pinout:
https://files.seeedstudio.com/products/102991455/WT32-ETH01_datasheet_V1.1-%20en.pdf

[image]

I also see the built in LED on the WT32 blinking when it is connected to the heat-pump:

[image]

However I constantly get this output:

[image]

Also tried using SoftwareSerial instead of HardwareSerial, which made no difference at all /:

Do you have any ideas on how to get this to run?
Thank you

pstarkl, 2022-12-22:
hello again!

I am finally getting data out of my daikin using the WT32 🥳 
the problem was the length of the dupont wires! 🙄 
I had connected 2 dunpont wires in a row for the pins to increase the length, but apperantly that causes the timeout!

closed 2022-12-22 by pstarkl

bwindsor, 2023-02-15:
@pstarkl Would you mind sharing how you swapped the WiFi connection with an Ethernet one please?

pstarkl, 2023-02-16:
@bwindsor Yeah sure, I will publish the code in a few days :)

Alibloke, 2024-02-14:
@pstarkl did you publish the code?

ivanleovitanza, 2024-11-30:
> @Alibloke did you find the code?