raomin/ESPAltherma #213: WiFi connection on Mesh WiFi system

TommasoPino, 2022-12-19:
Hi everyone, I am experiencing inconvenient behavior where the device that is connected with a high signal, for some reason, reset the wifi connection and it switches on a station that has a very low signal. I do not understand why it prefers to connect to a station of the mesh with a low signal, I think the reason is that there is no order to which station connects.

I have no time or experience in modifying the code, but hypothetically speaking using a library like this [https://randomnerdtutorials.com/esp32-wifimulti/](https://randomnerdtutorials.com/esp32-wifimulti/) could the problem be solved?

Thanks in advance for your time.

raomin, 2022-12-20:
I experience this too. The esp32 seems to randomly select the AP.
In my case both AP broadcast the same SSID so wifimulti would not help.
However, it seems that the problem is identified and is being fixed in esp-idf: https://github.com/espressif/esp-idf/issues/8269
(BTW it seems it's more an authentication failure (it can happen) on the best AP that make it end up on the worst).

Hopefully, it will solve this issue soon. 
Nothing related to ESPAltherma but I'll keep an eye on this to see if some code change is needed.

raomin, 2023-04-04:
I just pushed a commit that should fix this... 🤞

PR #343 (2024-01-05): Connect to strongest signal AP in case of multiple AP's with same SSID
Had issues with connecting to my WiFi network. Fixed this a while ago for my other project. Saw an earlier commit for #213 but it also being reverted a little later. This PR should fix these kind of issues for both ESP32 and ESP8266.

For ESP32 SDK based functions are used, tested with espressif32@6.4.0 and confirmed to be solving issues
For ESP8266 a function to find the AP with the strongest signal has been added, tested to be working.

Should fix #213 for both platforms
closed 2024-01-05 by raomin