raomin/ESPAltherma #224: M5Stick plus not able to connect anymore to wifi

gattigianluca, 2022-12-31:
I did all installation correctly and I was able to get data for half a day but then suddenly the M5Stick plus was not able to connect anymore to wifi.
What could be the reason? The wifi is not so good but it was able to send data for almost half a day.
Can I connect to M5Stcik via USB while it is still connected to the heat pump or it is better to disconnect from heat pump before doing any test? What is the correct way to remove the M5stick from heat pump: should I turn off the pump?
Thanks!

raomin, 2022-12-31:
Hi @gattigianluca, welcome.

What is the rssi value reported? (it's the wifi strength)
It could be that the wifi signal is deteriorated when the heat pump is in operation. Is it the case?
Also, for some reasons detailed [here](https://github.com/raomin/ESPAltherma/issues/213) the ESP might chose not the best access point when you have multiple AP broadcasting the same SSID...

gattigianluca, 2022-12-31:
> Hi @gattigianluca, welcome.
>
> What is the rssi value reported? (it's the wifi strength) #It could be that the wifi signal is deteriorated when the heat pump is in operation. Is it the case? Also, for [...]

The RSSI was not so good, i think something like -60, but it worked all night when pump was in operation.
I could change the SSID name of the AP where I have the pump so that ESP32 can not choose another one.
To remove the M5stick I should turn of the pump, correct?
Many thanks!

raomin, 2023-01-02:
> To remove the M5stick I should turn off the pump, correct?
Well, it's always safer to operate a main board when the machine is off.
You can also use OTA upgrade, so you don't even have to touch the esp.

gattigianluca, 2023-01-02:
> > To remove the M5stick I should turn off the pump, correct?
> 
> Well, it's always safer to operate a main board when the machine is off. You can also use OTA upgrade, so you don't even have to touch the esp.
> 
> I do understand that you were indeed in the situation where you have multiple AP broadcasting the same SSID, right?

Yes I changed the SSID for the AP in the basement, where ESP is located, and M5stick started to work again. At least so far :)
I am interested in the OTA upgrade, what is exactly? I would like to find a way to update ESPAlterma or change config without having to detouch every time!
Thanks!

raomin, 2023-01-02:
https://github.com/raomin/ESPAltherma#how-can-i-update-espaltherma-remotely
:)

gattigianluca, 2023-01-03:
> https://github.com/raomin/ESPAltherma#how-can-i-update-espaltherma-remotely :)

Quick question: how can I update the project from GIT before updating the M5stick via OTA?

Tardymo, 2023-01-05:
Stash your changed files (setup.h and definition header file). Use git stash command for that.
Pull the latest changes using git pull command.
Apply stashed changes back to working copy using git stash pop command
Build and upload

You can also perform these steps from within VS Code in Source Control section.

raomin, 2023-04-04:
FYI, the wifi selection in case of mesh should be fixed now.

closed 2023-04-04 by raomin