raomin/ESPAltherma #227: Understanding Voltage

AlgirdasPundzius, 2023-01-05:
First of all thanks for awesome tool !

I'm trying to do some additional calculations with the data (calculate COP, current wattage and etc)

But COP is showing crazy values. I spotted my voltage data is weird.
It fluctuates between 0 and 50. Though not always, during DHW cycle I had values of 330. So what does it exactly show ? Shouldn't voltage here be constant 220v ?

Using `{0x21,4,101,2,-1,"Voltage (N-phase) (V)"}, LT_DA_04-08KW`
My daikin model: ERGA04EAV3/EHVH04S18EA6V

Tardymo, 2023-01-09:
On my unit this value also changes from about 100V up to 400V, so it is of no use for me.

raomin, 2023-01-09:
Hi!
Yes, on some unit, voltage is not reported. Just use 220 in your calculation.

raomin, 2023-01-11:
Feel free to re-open if there is something new.

closed 2023-01-11 by raomin

domoneyleon, 2023-08-23:
@AlgirdasPundzius I have the same model heat pump as you and I was having trouble locating the x10a or the x12a serial port.  I was wondering if you could help me out.  I opened the case and searched the PCB for the port but couldn't see it.  In the 2nd image below I've circled the only free port that looks like an x10a/x12a port but I couldn't confirm from the angle I had.  Any help much appreciated!
[image]
[image]

domoneyleon, 2023-08-23:
I'm dealing with more than one heat pumps and both model numbers are ERGA04EAV3 and ERLA03DAV3